ansys/pyansys-geometry #739: PLOTTING: add affine transform widget

RobPasMue, 2023-09-19:
We should consider adding this new widget from PyVista https://github.com/pyvista/pyvista/pull/4803

AlejandroFernandezLuces, 2023-09-20:
Just FYI, this was added in `pyvista==0.42`, we still have the 0.41 version due to the documentation.

RobPasMue, 2023-09-20:
Okay, no problem :) Once it's available we will bump it. We will also have to bump the minimum PyVista requirement to 0.42 in that case

AlejandroFernandezLuces, 2023-09-22:
## Overview of this feature

### What we want

We were looking for an affine transform that applies the transformations to all of the scene. We were looking to substitute the axis widgets that we already have in the corner for an interactable one to move all of the objects of the scene around.

### What PyVista currently have

PyVista currently offers an affine transform widget, that is meant to be attached to a **particular actor*. This allows to apply the affine transformations to this object in particular. Currently, you can not perform affine transformations to a list of actors, which is what we were looking for.

### Conclusions

Since we cannot easily implement what we want with the current PyVista features, I think we should wait until this feature is implemented in PyVista to continue the development here.

RobPasMue, 2023-09-25:
Please open an issue on PyVista to see if they could implement an option to apply this widget to the entire scene. Thanks @AlejandroFernandezLuces!