ansys/pyansys-geometry #749: Bug located in `modeler.create_design`

kmahajan-cadfem, 2023-09-21:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

using local discovery as Backend(`launch_modeler_with_discovery()`) version 2023 R2, the function `modeler.create_design` does not work and since there is no design, it is not possible to anything with Discovery.

### 📝 Steps to reproduce

install the version `0.3.1`
code

```python
from ansys.geometry.core import launch_modeler_with_discovery
modeler = launch_modeler_with_discovery(product_version=232)
design = modeler.create_design(name="des")
```

output

```shell
---------------------------------------------------------------------------
GeometryExitedError                       Traceback (most recent call last)
Cell In[3], line 1
----> 1 design = modeler.create_design(name="des")

File <path-to-env>\lib\site-packages\ansys\geometry\core\modeler.py:143, in Modeler.create_design(self, name)
    140 from ansys.geometry.core.designer.design import Design
    142 check_type(name, str)
--> 143 design = Design(name, self._client)
    144 self._designs.append(design)
    145 if len(self._designs) > 1:

File <path-to-env>\lib\site-packages\ansys\geometry\core\errors.py:101, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     99     out = func(*args, **kwargs)
    100 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
--> 101     raise GeometryExitedError(
    102         f"Geometry service connection terminated: {error.details()}"
    103     ) from None
    105 if threading.current_thread().__class__.__name__ == "_MainThread":
    106     received_interrupt = bool(SIGINT_TRACKER)

GeometryExitedError: Geometry service connection terminated: Command not found.
Parameter name: commandName
Command not found.
Parameter name: commandName
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2023 R2

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansys.geometry.core==0.3.1
```

RobPasMue, 2023-09-21:
Hi @kmahajan-cadfem! Thanks for opening this issue.

First question, is the connection healthy with Discovery? And, does the Discovery GUI show up?

You can do this as follows:

```py
from ansys.geometry.core import launch_modeler_with_discovery

modeler = launch_modeler_with_discovery(product_version=232)
print(modeler)
```

kmahajan-cadfem, 2023-09-21:
@RobPasMue, it is healthy without any problems and the UI shows up

```python
print(modeler)
```
-------------

Output

```shell
Ansys Geometry Modeler (0x1f90f6cd240)

Ansys Geometry Modeler Client (0x1f90f6cdf90)
  Target:     127.0.0.1:57522
  Connection: Healthy

```

I thought this might be something in wrapper, so I have also tried following but it did not help

```python
from ansys.api.geometry.v0.designs_pb2_grpc import DesignsStub
from ansys.api.geometry.v0.designs_pb2 import ExportRequest, NewRequest, SaveAsRequest
_design_stub = DesignsStub(modeler.client.channel)
new_design = _design_stub.New(NewRequest(name="53"))
print(new_design.id)
```

I get following output

```shell
---------------------------------------------------------------------------
_InactiveRpcError                         Traceback (most recent call last)
Cell In[21], line 4
      2 from ansys.api.geometry.v0.designs_pb2 import ExportRequest, NewRequest, SaveAsRequest
      3 _design_stub = DesignsStub(modeler.client.channel)
----> 4 new_design = _design_stub.New(NewRequest(name="53"))
      5 print(new_design.id)

File <path-to-venv>\lib\site-packages\grpc\_channel.py:1161, in _UnaryUnaryMultiCallable.__call__(self, request, timeout, metadata, credentials, wait_for_ready, compression)
   1146 def __call__(
   1147     self,
   1148     request: Any,
   (...)
   1153     compression: Optional[grpc.Compression] = None,
   1154 ) -> Any:
   1155     (
   1156         state,
   1157         call,
   1158     ) = self._blocking(
   1159         request, timeout, metadata, credentials, wait_for_ready, compression
   1160     )
-> 1161     return _end_unary_response_blocking(state, call, False, None)

File <path-to-venv>\lib\site-packages\grpc\_channel.py:1004, in _end_unary_response_blocking(state, call, with_call, deadline)
   1002         return state.response
   1003 else:
-> 1004     raise _InactiveRpcError(state)

_InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Command not found.
Parameter name: commandName
Command not found.
Parameter name: commandName"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-09-21T09:00:15.9668242+00:00", grpc_status:2, grpc_message:"Command not found.\r\nParameter name: commandName\nCommand not found.\r\nParameter name: commandName"}"
>
```

Then I did take a look at the proto files from [api](https://github.com/ansys/ansys-api-geometry) repo but seems like it is going to change anyways in future a lot because release `0.3.*` from  [api](https://github.com/ansys/ansys-api-geometry) does not have designs.proto at all !?

RobPasMue, 2023-09-21:
You are right, starting on 24.1 the proto files will change significantly. Right now, ``0.2.10`` is the one that ensures compatibility with 23.2 but as you may see... there is no possibility to launch Discovery or SpaceClaim from that package version. That is only available from ``0.3.*`` onwards.

Support for PyAnsys Geometry with the Ansys Products (SpaceClaim/Discovery) is limited on 23.2. For operating with 23.2 I recommend you use either SpaceClaim or the Geometry Service directly, rather than Discovery. Its launch methods are very similar.

``launch_modeler_with_spaceclaim``
``launch_modeler_with_geometry_service`` --> There is a bug in the docs, this one does not accept ``product_version``. We will fix this asap.

You do have 23R2.1 installed, right? This is only available from SP1 onwards

RobPasMue, 2023-09-21:
Also, I suggest that you give it a try to the Dockerized version of the service 😃 the binaries are available on the release page, and the Docker container is known to work for sure - we heavily test it. These Docker containers are using the Geometry Service behind the scenes so a local launch of the Geometry Service should also work fine (i.e. ``launch_modeler_with_geometry_service``)

kmahajan-cadfem, 2023-09-21:
> You are right, starting on 24.1 the proto files will change significantly. Right now, `0.2.10` is the one that ensures compatibility with 23.2 but as you may see... there is no possibility to launch [...]

Thanks for the information,
I did install SP1 explicitly to test this one out,  
FYI, It does work without problem in SCDM, already tested that.
since this would change, we will wait at least until we get our hands on 24.1 installation.
sadly, Dockerized version is not yet an option for me. 

you can decide, what to do with the issue

RobPasMue, 2023-09-21:
Let me try to get back to you this afternoon with a final response from the rest of the team =) thanks again for your feedback @kmahajan-cadfem! I haven't forgot about the units issue you opened. I'm planning work on it fairly soon.

RobPasMue, 2023-09-21:
Hi @kmahajan-cadfem - I managed to get in touch with the rest of the team... This issue won't be fixed on Discovery 23.2 - since it depends on the product's implementation. Discovery support will thus be available from 24.1 onwards. Sorry for the inconveniences! I will close the issue - but feel free to post any further comments.

closed 2023-09-21 by RobPasMue

RobPasMue, 2023-09-21:
And that's right - priority is set for Discovery, and SCDM will be deprecated in the future.

I just merged #750 which will try to clarify it. In the meantime, even though Discovery is not yet available, let me clarify some things:

* PyAnsys Geometry came up as a Python Client for the Ansys Geometry Service - a headless solution based on the underlying libraries of Discovery and SpaceClaim.
* Support for the GUI products (SpaceClaim and Discovery), at least on this first release, is on a best effort basis. Once 24.1 comes out we expect to be compatible with both of them fully.
* Even though the Discovery interaction for 23.2 is not working (which again, I am really sorry) - I want to emphasize the fact that the PyAnsys Geometry scripts will look the same irrespectively of the backend you are connecting to right now for your modelling operations. Meaning that you could develop your script right now using any of the options available (Geometry Service, Dockerized version or SpaceClaim) and the output should be the same.

Again, I understand your concerns for 23R2 and Discovery, but I would encourage you to still try it out, because whatever script you come up with based on the PyAnsys Geometry API will still be runnable with any of the backends starting on 24.1 (or at least that's the goal! 😄 )

kmahajan-cadfem, 2023-09-21:
@RobPasMue, got it, I was creating some demo material and wanted to show the PyAnsys Geometry, that is the reason I wanted to show Discovery, and at least for initial try-outs would also be customers first choice, I think.
I will give it a try to use the headless version here onward 👍🏻

RobPasMue, 2023-09-21:
Yes definitely! We understand that whenever you are initially working on a script/workflow (in which you are not interested on performance - but on getting results) using the connection with Discovery would be great! Because you would see the changes directly on screen 😃 

It is only once you have a robust script that you would go on headless mode - I get it =)

Give it a try to headless, and try using our visualization tool (based on PyVista). We have some examples on our documentation! I know it won't be in the same but at least it will allow you to play with it in the meantime!

Thanks again for being one of our first users! ❤️