ansys/pyansys-geometry #756: Bug located in `launch_modeler_with_geometry_service`

kmahajan-cadfem, 2023-09-25:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

using following code does not work

```python
from ansys.geometry.core import launch_modeler_with_geometry_service

modeler_discovery = launch_modeler_with_geometry_service()
```

```shell
TimeoutError                              Traceback (most recent call last)
Cell In[6], line 3
      1 from ansys.geometry.core import launch_modeler_with_geometry_service
----> 3 modeler_discovery = launch_modeler_with_geometry_service()

File <path-to-venv>\lib\site-packages\ansys\geometry\core\connection\launcher.py:349, in launch_modeler_with_geometry_service(host, port, enable_trace, log_level, timeout)
    279 def launch_modeler_with_geometry_service(
    280     host: str = "localhost",
    281     port: int = None,
   (...)
    284     timeout: int = 60,
    285 ) -> "Modeler":
    286     """
    287     Start the Geometry service locally using the ``ProductInstance`` class.
    288 
   (...)
    347         timeout=300)
    348     """
--> 349     return prepare_and_start_backend(
    350         BackendType.WINDOWS_SERVICE,
    351         host=host,
    352         port=port,
    353         enable_trace=enable_trace,
    354         log_level=log_level,
    355         api_version=ApiVersions.LATEST,
    356         timeout=timeout,
    357     )

File <path-to-venv>\lib\site-packages\ansys\geometry\core\connection\product_instance.py:240, in prepare_and_start_backend(backend_type, product_version, host, port, enable_trace, log_level, api_version, timeout)
    236 LOG.debug(f"Environment variables: {env_copy}")
    238 instance = ProductInstance(_start_program(args, env_copy).pid)
--> 240 return Modeler(
    241     host=host, port=port, timeout=timeout, product_instance=instance, backend_type=backend_type
    242 )

File <path-to-venv>\lib\site-packages\ansys\geometry\core\modeler.py:105, in Modeler.__init__(self, host, port, channel, remote_instance, local_instance, product_instance, timeout, logging_level, logging_file, backend_type)
     91 def __init__(
     92     self,
     93     host: str = DEFAULT_HOST,
   (...)
    102     backend_type: Optional[BackendType] = None,
    103 ):
    104     """Initialize the ``Modeler`` class."""
--> 105     self._client = GrpcClient(
    106         host=host,
    107         port=port,
    108         channel=channel,
    109         remote_instance=remote_instance,
    110         local_instance=local_instance,
    111         product_instance=product_instance,
    112         timeout=timeout,
    113         logging_level=logging_level,
    114         logging_file=logging_file,
    115         backend_type=backend_type,
    116     )
    118     # Design[] maintaining references to all designs within the modeler workspace
    119     self._designs = []

File <@beartype(ansys.geometry.core.connection.client.GrpcClient.__init__) at 0x1fa1fb5ac20>:234, in __init__(__beartype_func, __beartype_conf, __beartype_get_violation, __beartype_object_2173783550656, __beartype_object_2173784530496, __beartype_object_2173784536832, __beartype_object_140723746898672, __beartype_typistry, __beartype_object_2175375122560, __beartype_object_2175322714496, __beartype_object_2173784563424, __beartype_object_2173784273280, __beartype_object_2173784367680, __beartype_object_2173784263872, *args, **kwargs)

File <path-to-venv>\lib\site-packages\ansys\geometry\core\connection\client.py:156, in GrpcClient.__init__(self, host, port, channel, remote_instance, local_instance, product_instance, timeout, logging_level, logging_file, backend_type)
    148     self._channel = grpc.insecure_channel(
    149         self._target,
    150         options=[
    151             ("grpc.max_receive_message_length", MAX_MESSAGE_LENGTH),
    152         ],
    153     )
    155 # do not finish initialization until channel is healthy
--> 156 wait_until_healthy(self._channel, timeout)
    158 # once connection with the client is established, create a logger
    159 self._log = logger.add_instance_logger(
    160     name=self._target, client_instance=self, level=logging_level
    161 )

File <path-to-venv>\lib\site-packages\ansys\geometry\core\connection\client.py:77, in wait_until_healthy(channel, timeout)
     75 else:
     76     target_str = channel._channel.target().decode()
---> 77     raise TimeoutError(
     78         f"Channel health check to target '{target_str}' timed out after {timeout} seconds."
     79     )

TimeoutError: Channel health check to target 'localhost:64638' timed out after 60 seconds.

```

I have also tried to start the `Modeling Service` without the pygeometry but it does not start

```shell
C:/Program Files/ANSYS Inc/v232/GeometryServices/Presentation.ApiServerDMS.exe
```
but i get the following output

```shell
C:/Program Files/ANSYS Inc/v232/GeometryServices/Presentation.ApiServerDMS.exe"

Unhandled Exception: System.UnauthorizedAccessException: Access to the path 'C:\Program Files\ANSYS Inc\v232\GeometryServices\Logs' is denied.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.Directory.InternalCreateDirectory(String fullPath, String path, Object dirSecurityObj, Boolean checkHost)
   at System.IO.Directory.InternalCreateDirectoryHelper(String path, Boolean checkHost)
   at Infrastructure.ApiGeometry.Utilities.SpecificImplementationsService.InstantiateApi()
   at Infrastructure.ApiGeometry.ApiGeometry.get_Instance()
   at Infrastructure.ApiGeometryProvider.Utilities.SpecificImplementation.GetApi(ApiGeometryVersionsCatalog version)
   at Infrastructure.ApiGeometryProvider.ApiGeometryProvider.SelectApiVersion(ApiGeometryVersionsCatalog apiVersion)
   at Infrastructure.ApiGeometryProvider.ApiGeometryProvider..ctor()
   at Infrastructure.ApiGeometryProvider.ApiGeometryProvider.get_Instance()
   at Presentation.ApiServerDMS.ApiServer.StartServer()
   at Presentation.ApiServerDMS.ApiServer.Main(String[] args)
```

I can understand the error, unfortunately I cannot change it, we do not have write access on the `C:/` drive, which is also very common in many cases for customers as well.

since this looks like limitation for the server, it would be great to at least change the docs for this as well and for next version(241) change the log position somewhere else, or better give the user(PyAnsys) choice over the arguments.

if this is already fixed in the newest builds please ignore and close this issue

### 📝 Steps to reproduce

```shell
C:/Program Files/ANSYS Inc/v232/GeometryServices/Presentation.ApiServerDMS.exe
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

232

### 🐍 Which Python version are you using?

3.7

### 📦 Installed packages

```shell
-
```

kmahajan-cadfem, 2023-09-25:
@RobPasMue - Thanks for the detailed explanation, I was not sure if I can copy paste the folder or not but since you have mentioned it, I will try it later, and give a feedback. 👍🏻

RobPasMue, 2023-09-25:
>  I was not sure if I can copy paste the folder or not 

It's not intended... but worth a try 😄 Let me know if it works!

kmahajan-cadfem, 2023-09-25:
It did work without problems 👍🏻  and can also connect to service.

RobPasMue, 2023-09-25:
Well, at least you got a quick fix 😄 

We will continue to improve it - we will provide users with the option to redefine the location of the Logs directory though PyAnsys Geometry for 241.

RobPasMue, 2023-10-27:
@kmahajan-cadfem - once #831 is fixed you should be able to redefine the logs directory =)

closed 2023-10-27 by RobPasMue