ansys/pyansys-geometry #786: Reactivate Linux testing

RobPasMue, 2023-10-04:
Due to licensing issues, the Linux tests were deactivated. This task is planned to reactivate them once fixed. Related to #785

kmahajan-cadfem, 2023-10-06:
@RobPasMue: does this mean, the linux support has been completed for the DMS, I also found [this ](https://geometry.docs.pyansys.com/version/dev/getting_started/docker/linux_container.html), can we try it out, or  are there still some open issues regarding the linux support?

RobPasMue, 2023-10-06:
Hi @kmahajan-cadfem - the Linux version of the Geometry service is available for beta usage. Many features are not available yet,  and they won't be available for some time I presume... But you can build geometries, save to file and so on. We do not have a list of the supported APIs yet so my suggestion is that you give it a try in the meantime or that if you have any doubts, you ping us.

However, the binaries to build the container are no longer usable (the ones on the release page). We are preparing for a new release candidate (0.4.r2) that will re-enable its usage. I'm in process of getting that release candidate now in fact, some time today

kmahajan-cadfem, 2023-10-06:
Thanks for the detailed info, its not super urgent, but I will keep you posted, 

some background:
since the current installation is not working out of the box we would like to create an automation where we only use the `run_script` and docker if that is possible, and for me its easier to use the docker on linux instead of windows.
do you think there would be any problems for only `run_script`?

RobPasMue, 2023-10-06:
``run_script`` is definitely not available on the Linux container, I'm sorry to inform you of that... That feature is indeed one of the most complex to port over to Linux (from what the service dev team has always told me)... I doubt it will ever be available.

kmahajan-cadfem, 2023-10-06:
@RobPasMue: thank you for the heads up and glad that I cleared it before we work on something. 
I will have to think of some other strategy at least until the 24 R1 is available, as per your comment the DMS is available as binaries with which we can create the images, is there any official page where it is available to download like [DPF](https://download.ansys.com/Others/DPF%20Pre-Release) ?

RobPasMue, 2023-10-06:
For now, the official page we have is the GitHub Releases page for each PyAnsys Geometry version. You can follow the instructions here: https://geometry.docs.pyansys.com/version/dev/getting_started/docker/windows_container.html#build-the-geometry-service-windows-container

We are planning on adding ourselves to the same strategy as DPF, but for now it is not available.

kmahajan-cadfem, 2023-10-06:
Thank you very much for the quick replies

RobPasMue, 2023-10-06:
Anytime @kmahajan-cadfem =)

PR #796 (2023-10-06): cicd: reacitvate Linux testing
Closes #786