ansys/pyansys-geometry #788: Add support for Python 3.12

RobPasMue, 2023-10-04:
Related to #787 - we tried it but ``vtk`` was blocking the update.

ChristosT, 2023-11-02:
Great to see interest for python 3.12 so early ! The latest [rc](https://pypi.org/project/vtk/9.3.0rc2/ )  comes with python 3.12 support. Could you give it a try and report related issues at
https://gitlab.kitware.com/vtk/vtk/-/issues ?

RobPasMue, 2023-11-03:
Okay sounds like it "works" :)

We are only doing smoke testing - checking that the package builds and that we are able to import its version. Our main dev versions are Python 3.9 to 3.11. This is just to provide support for Python 3.12 on a best-effort basis. Thanks @ChristosT for pointing out the current 9.0.3rc2 release.

I am curious... when is the actual 9.3.0 release expected?

ChristosT, 2023-11-03:
That's hard to tell . Depends on how the rc goes and if any issues are reported. If everything goes well the final release should be in a few weeks.

MaxJPRey, 2023-11-03:
Thanks @RobPasMue for taking care of this first set of testing. It will be halpful for all our other repos.

ChristosT, 2023-11-14:
> I am curious... when is the actual 9.3.0 release expected?

vtk 9.3.0 is available on [pypi](https://pypi.org/project/vtk/)

RobPasMue, 2023-11-15:
Yep!! Dependabot was so kind that opened #854 for us =) Thanks @ChristosT!

PR #789 (2023-11-15): feat: adding Python 3.12 support
Closes #788
closed 2023-11-15 by RobPasMue