ansys/pyansys-geometry #801: IGES import on Windows service provides failing results

RobPasMue, 2023-10-07:
See TODO in  ``test_design_import.py::test_open_file`` - latest DMS version has started to fail here. Requires service investigation. Assigning to @jonahrb.

RobPasMue, 2023-10-07:
Read lines

https://github.com/ansys/pyansys-geometry/blob/114d4e82163161553b71a678a95eecf80be782bc/tests/integration/test_design_import.py#L168-L174

jonahrb, 2023-10-09:
This is a known issue of additional empty components that occurs in SC product as well. Likely a stride bug, it has been reported. So this IGES test won't pass as-is until that is fixed. Other option as a possible workaround is to explicitly ignore empty components, but that's probably not ideal? @b-matteo @kmental , feel free to add any additional comments

b-matteo, 2023-10-09:
> This is a known issue of additional empty components that occurs in SC product as well. Likely a stride bug, it has been reported. So this IGES test won't pass as-is until that is fixed. Other [...]

Thanks for the heads up @jonahrb . Personnally i think it's more consistent to keep the test the way it is and wait for a fix. If the fix is to be too delayed, we could make a specific test for IGES.

closed 2024-04-30 by RobPasMue