ansys/pyansys-geometry #809: Bug located in `run_discovery_script_file`

kmahajan-cadfem, 2023-10-13:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

I always get the error that script failed

```
GeometryRuntimeError: Script run failed: An error occurred: One or more errors occurred.Stack trace:
```

### 📝 Steps to reproduce

- launch DMS (because of #756)
- use following to connect to the modeler

```python
from ansys.geometry.core import Modeler
modeler = Modeler(port=50051)
```

- use [this file](https://github.com/ansys/pyansys-geometry/blob/9892e2b79bcb32293eb191a8dffb446a1172119b/tests/integration/files/disco_scripts/simple_script.py)

- run script using

```python
modeler.run_discovery_script_file("test_disc.py",{})
```

- it does not matter which script you use, even just one line script does not work. 

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2023 R2 (DMS on D drive)

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansys.geometry.core==0.3.2
```

kmahajan-cadfem, 2023-10-13:
@RobPasMue, do you have any thoughts or maybe a workaround?

RobPasMue, 2023-10-15:
Let me check this on Monday morning and I'll get back to you! =)

kmahajan-cadfem, 2023-10-18:
@RobPasMue, I am sorry to keep you bothering, did you had a chance to take a look, we are in process of feasibility for a potential customer, if you can give me some workaround that would be great.

b-matteo, 2023-10-18:
@RobPasMue @kmahajan-cadfem, sure, i'll take it from here.
I'm following the steps to reproduce and let you know as soon as i find something.

kmahajan-cadfem, 2023-10-18:
Thanks @RobPasMue  and @b-matteo

kmahajan-cadfem, 2023-10-19:
@b-matteo, did you find something in this case, would it work with version or we have to wait until the next version to test out the workflow, it would also be sufficient for the moment if we get the artifact that is mentioned [here](https://geometry.docs.pyansys.com/version/dev/getting_started/docker/windows_container.html#prerequisites).

b-matteo, 2023-10-19:
Hi @kmahajan-cadfem , it looks like the DMS in 23.2.2 is having issues with the run_discovery_script_file in release. Possible workarounds would be having ANSYS SpaceClaim232 as backend instead, or switching to Ansys GeometryService (DMS) 24.1 and PyAnsys-Geometry 0.4 as soon as available.

b-matteo, 2023-10-19:
No problem, apologies for the trouble here.

closed 2023-10-19 by RobPasMue