ansys/pyansys-geometry #834: Surface bodies are not recognized.

umutsoysalansys, 2023-10-31:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

when a design is imported ( tests\integration\files\DuplicateFacesDesignBefore.scdocx) it is seen that there is 1 body and 1 surface body on the SC. However they both are represented as non surface body on the client side. 
[image]

### 📝 Steps to reproduce

Read the design, print bodies. 

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

SpaceClaim V241.

### 🐍 Which Python version are you using?

3.8

### 📦 Installed packages

```shell
accessible-pygments==0.0.4
aiohttp==3.8.5
aiosignal==1.3.1
alabaster==0.7.13
ansys-api-dbu==0.2.0
ansys-api-geometry==0.3.0
ansys-api-platform-instancemanagement==1.0.0
-e git+https://github.com/ansys/pyansys-geometry.git@607650079d49177a334ec1ffee03c585de6c52b3#egg=ansys_geometry_core
ansys-platform-instancemanagement==1.1.2
ansys-sphinx-theme==0.12.1
ansys-tools-path==0.3.1
anyascii==0.3.2
anyio==4.0.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
astroid==3.0.0
asttokens==2.4.0
async-lru==2.0.4
async-timeout==4.0.3
attrs==23.1.0
Babel==2.12.1
backcall==0.2.0
beartype==0.16.2
beautifulsoup4==4.12.2
bleach==6.0.0
bokeh==3.2.2
cachetools==5.3.1
certifi==2023.7.22
cffi==1.16.0
cfgv==3.4.0
chardet==5.2.0
charset-normalizer==3.3.0
click==8.1.7
codecov-cli==0.3.8
colorama==0.4.6
comm==0.1.4
contourpy==1.1.1
coverage==7.3.1
cycler==0.12.0
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
distlib==0.3.7
docker==6.1.3
docutils==0.20.1
exceptiongroup==1.1.3
executing==2.0.0
fastjsonschema==2.18.1
filelock==3.12.4
fonttools==4.43.0
fqdn==1.5.1
frozenlist==1.4.0
google-api-core==2.12.0
google-api-python-client==2.101.0
google-auth==2.23.2
google-auth-httplib2==0.1.1
googleapis-common-protos==1.60.0
grpcio==1.50.0
grpcio-health-checking==1.48.2
h11==0.14.0
httpcore==0.16.3
httplib2==0.22.0
httpx==0.23.3
identify==2.5.30
idna==3.4
ijson==3.2.3
imagesize==1.4.1
importlib-metadata==6.8.0
iniconfig==2.0.0
ipycanvas==0.13.1
ipyevents==2.0.2
ipykernel==6.25.2
ipython==8.16.1
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.2
json5==0.9.14
jsonpointer==2.4
jsonschema==4.19.1
jsonschema-specifications==2023.7.1
jupyter-events==0.7.0
jupyter-lsp==2.2.0
jupyter-sphinx==0.4.0
jupyter_client==8.3.1
jupyter_core==5.3.2
jupyter_server==2.7.3
jupyter_server_terminals==0.4.4
jupyterlab==4.0.6
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.7
jupyterlab_server==2.25.0
jupytext==1.15.2
kiwisolver==1.4.5
linkify-it-py==2.0.2
Markdown==3.4.4
markdown-it-py==3.0.0
MarkupSafe==2.1.3
matplotlib==3.8.0
matplotlib-inline==0.1.6
mdit-py-plugins==0.4.0
mdurl==0.1.2
mistune==3.0.2
multidict==6.0.4
myst-parser==2.0.0
nbclient==0.8.0
nbconvert==7.8.0
nbformat==5.9.2
nbsphinx==0.9.3
nest-asyncio==1.5.8
nodeenv==1.8.0
notebook==7.0.4
notebook_shim==0.2.3
numpy==1.26.0
numpydoc==1.6.0
overrides==7.4.0
packaging==23.2
pandas==2.1.1
pandocfilters==1.5.0
panel==1.2.3
param==1.13.0
parso==0.8.3
pickleshare==0.7.5
Pillow==10.0.1
Pint==0.22
platformdirs==3.11.0
pluggy==1.3.0
pooch==1.7.0
pre-commit==3.4.0
prometheus-client==0.17.1
prompt-toolkit==3.0.39
protobuf==3.20.3
psutil==5.9.5
pure-eval==0.2.2
pyasn1==0.5.0
pyasn1-modules==0.3.0
pycparser==2.21
pydata-sphinx-theme==0.14.1
Pygments==2.16.1
pyparsing==3.1.1
pyproject-api==1.6.1
pytest==7.4.2
pytest-cov==4.1.0
pytest-xvfb==3.0.0
pytest_pyvista==0.1.9
python-dateutil==2.8.2
python-json-logger==2.0.7
pytz==2023.3.post1
PyVirtualDisplay==3.0
pyvista==0.41.1
pyviz_comms==3.0.0
pywin32==306
pywinpty==2.0.11
PyYAML==6.0.1
pyzmq==25.1.1
referencing==0.30.2
requests==2.31.0
responses==0.21.0
rfc3339-validator==0.1.4
rfc3986==1.5.0
rfc3986-validator==0.1.1
rpds-py==0.10.3
rsa==4.9
scipy==1.11.3
scooby==0.7.4
Send2Trash==1.8.2
six==1.16.0
smart-open==6.4.0
sniffio==1.3.0
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.2.5
sphinx-autoapi==3.0.0
sphinx-autodoc-typehints==1.24.0
sphinx-copybutton==0.5.2
sphinx-jinja==2.0.2
sphinx_design==0.5.0
sphinxcontrib-applehelp==1.0.7
sphinxcontrib-devhelp==1.0.5
sphinxcontrib-htmlhelp==2.0.4
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.6
sphinxcontrib-serializinghtml==1.1.9
stack-data==0.6.3
tabulate==0.9.0
terminado==0.17.1
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tornado==6.3.3
tox==4.11.3
tqdm==4.66.1
traitlets==5.10.1
trame==3.2.7
trame-client==2.12.0
trame-server==2.12.0
trame-vtk==2.5.8
trame-vuetify==2.3.1
tree-sitter==0.20.2
types-python-dateutil==2.8.19.14
typing_extensions==4.8.0
tzdata==2023.3
uc-micro-py==1.0.2
uri-template==1.3.0
uritemplate==4.1.1
urllib3==2.0.5
virtualenv==20.24.5
vtk==9.2.6
wcwidth==0.2.8
webcolors==1.13
webencodings==0.5.1
websocket-client==1.6.3
widgetsnbextension==3.6.6
wslink==1.12.2
xyzservices==2023.7.0
yarl==1.9.2
zipp==3.17.0
```

RobPasMue, 2023-11-02:
Interesting, looks like on import this information is not stored. @AlejandroFernandezLuces can you investigate this?

AlejandroFernandezLuces, 2023-11-02:
I was looking into this and I think we are missing some information from the server side to fix this. The `body` response from the server doesn't contain any information telling if a body is a surface or not. 

In any case, I think it can be inferred that a body is a surface if it only has two faces and both have a planar surface type, but I think it would be more robust if the information comes directly from the server side.

There is a TODO regarding this in the code: https://github.com/ansys/pyansys-geometry/blob/29cc9e5fb475474c665f2caa2fc79a1a9ef9aa63/src/ansys/geometry/core/designer/design.py#L679-L684

@jonahrb could you give us your feedback on this? Does this need to be implemented on the server side?

jonahrb, 2023-11-03:
@AlejandroFernandezLuces I think you're right that we should just include that in the grpc body message. Surface bodies can take many forms, so it's best not to try to deduce it on the client side, let's just send it back from the server.

RobPasMue, 2023-11-03:
Agreed, let's do that. Could you work on that server-side @jonahrb? =)

jonahrb, 2023-11-06:
@RobPasMue @AlejandroFernandezLuces PR is up server-side. Once merged you can handle it on client

RobPasMue, 2023-11-06:
Thanks @jonahrb - great to hear that!

PR #846 (2023-11-20): feat: import surface body properly when reading design + boolean ops with multiple bodies
Closes #834
closed 2023-11-20 by RobPasMue