ansys/pyansys-geometry #897: Bug located in ...

XIAO-GUO-commint, 2023-12-28:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

My question is How to enlarge an image and center it?

from ansys.geometry.core import launch_modeler_with_spaceclaim
modeler_discovery = launch_modeler_with_spaceclaim()
modeler_discovery.run_discovery_script_file( file_path=r'E:\RuanZhu-files\Optimization_Muti\parameters_modeling\1.scscript')
design=modeler_discovery.read_existing_design()
design.plot()

[image]

### 📝 Steps to reproduce

from ansys.geometry.core import launch_modeler_with_spaceclaim
modeler_discovery = launch_modeler_with_spaceclaim()
modeler_discovery.run_discovery_script_file( file_path=r'E:\RuanZhu-files\Optimization_Muti\parameters_modeling\1.scscript')
design=modeler_discovery.read_existing_design()
design.plot()

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

ansys2024r1

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
Package                               Version
------------------------------------- ---------------
about-time                            4.2.1
aiohttp                               3.9.1
aiosignal                             1.3.1
alembic                               1.12.0
alive-progress                        3.1.4
altgraph                              0.17.4
ansys-api-dbu                         0.2.2
ansys-api-fluent                      0.3.18
ansys-api-geometry                    0.3.5
ansys-api-mapdl                       0.5.1
ansys-api-platform-instancemanagement 1.0.0
ansys-fluent-core                     0.18.0
ansys-fluent-visualization            0.8.0
ansys-geometry-core                   0.4.3
ansys-mapdl-core                      0.67.0
ansys-mapdl-reader                    0.52.20
ansys-math-core                       0.1.3
ansys-platform-instancemanagement     1.1.2
ansys-tools-path                      0.3.2
anyio                                 4.0.0
appdirs                               1.4.4
argon2-cffi                           23.1.0
argon2-cffi-bindings                  21.2.0
arrow                                 1.3.0
asttokens                             2.4.0
async-lru                             2.0.4
attrs                                 23.1.0
autograd                              1.6.2
[... 175 lines cut ...]
tornado                               6.3.3
tqdm                                  4.66.1
traitlets                             5.11.2
trame                                 3.3.0
trame-client                          2.13.0
trame-server                          2.12.1
types-python-dateutil                 2.8.19.14
typing_extensions                     4.8.0
tzdata                                2023.3
uri-template                          1.3.0
uritemplate                           4.1.1
urllib3                               2.0.7
vtk                                   9.3.20230807rc0
wcwidth                               0.2.8
webcolors                             1.13
webencodings                          0.5.1
websocket-client                      1.6.4
wheel                                 0.41.2
widgetsnbextension                    4.0.9
wrapt                                 1.15.0
wslink                                1.12.4
xgboost                               2.0.0
yarl                                  1.9.4
zipp                                  3.17.0
```

RobPasMue, 2024-01-02:
@AlejandroFernandezLuces please answer this user.

XIAO-GUO-commint, 2024-01-03:
Thank you very much! Both of your methods are effective, but it seems that the second method cannot open the PyVista interface in Jupyter Notebook.

AlejandroFernandezLuces, 2024-01-03:
We are currently working on the Jupyter Notebook visualization, since it has recently changed in PyVista and we are still adapting to it. Sorry for the inconvenience and we'll let you know once this is implemented.

AlejandroFernandezLuces, 2024-01-05:
> Thank you very much! Both of your methods are effective, but it seems that the second method cannot open the PyVista interface in Jupyter Notebook.

This was just fixed in PR #904 , to use it you can either use the dev version of the package or wait for the next release of the package. Closing this issue as it is solved, let me know if it works for you if you try it out 🙂

closed 2024-01-05 by AlejandroFernandezLuces