ansys/pyansys-geometry #947: Modeler.open_file()

Alberto-DM, 2024-01-23:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

Opening the attached file, results in the error `'J_Kg_Kdeg' is not defined in the unit registry.`. 
Discovery 24R1 can open the file just fine, without errors.
If we use Discovery app as local service, we can see that the document is opened without errors, but the `open_file` method throws the Exception. 
[DOORS_081202.BUG.zip](https://github.com/ansys/pyansys-geometry/files/14021461/DOORS_081202.BUG.zip)

### 📝 Steps to reproduce

```
from ansys.geometry.core import launch_modeler_with_discovery
from ansys.geometry.core.misc.options import ImportOptions

cad_filename = 'DOORS_081202.BUG.dsco'

modeler = launch_modeler_with_discovery(product_version=241)

options = ImportOptions()
options.cleanup_bodies = False
options.import_coordinate_systems = False
options.import_curves = False
options.import_hidden_components_and_geometry = False
options.import_names = False
options.import_planes = False
options.import_points = False
design = modeler.open_file(file_path=cad_filename, upload_to_server=False, import_options=options)
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys 24R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansi2html==1.9.1
ansys-api-dbu==0.2.2
ansys-api-geometry==0.3.5
ansys-geometry-core==0.4.4
ansys-tools-path==0.4.1
asttokens==2.4.1
attrs==23.2.0
beartype==0.16.4
black==23.12.1
blinker==1.7.0
cachetools==5.3.2
certifi==2023.11.17
cgal==5.6.post202312191118
charset-normalizer==3.3.2
click==8.1.7
colorama==0.4.6
comm==0.2.1
ConfigArgParse==1.7
contourpy==1.2.0
cycler==0.12.1
dash==2.14.2
dash-core-components==2.0.0
dash-html-components==2.0.0
dash-table==5.0.0
decorator==5.1.1
exceptiongroup==1.2.0
executing==2.0.1
fastjsonschema==2.19.1
Flask==3.0.0
fonttools==4.47.2
google-api-core==2.15.0
google-api-python-client==2.113.0
google-auth==2.26.2
google-auth-httplib2==0.2.0
googleapis-common-protos==1.62.0
grpcio==1.60.0
grpcio-health-checking==1.60.0
httplib2==0.22.0
idna==3.6
importlib-metadata==7.0.1
ipython==8.20.0
ipywidgets==8.1.1
itsdangerous==2.1.2
jedi==0.19.1
Jinja2==3.1.3
jsonschema==4.21.0
jsonschema-specifications==2023.12.1
jupyter_core==5.7.1
jupyterlab-widgets==3.0.9
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.8.2
matplotlib-inline==0.1.6
mypy-extensions==1.0.0
nbformat==5.9.2
nest-asyncio==1.5.9
numpy==1.26.3
open3d==0.18.0
packaging==23.2
parso==0.8.3
pathspec==0.12.1
pillow==10.2.0
Pint==0.23
platformdirs==4.1.0
plotly==5.18.0
pooch==1.8.0
prompt-toolkit==3.0.43
protobuf==4.25.2
pure-eval==0.2.2
pyasn1==0.5.1
pyasn1-modules==0.3.0
Pygments==2.17.2
pyparsing==3.1.1
python-dateutil==2.8.2
pyvista==0.43.2
pywin32==306
referencing==0.32.1
requests==2.31.0
retrying==1.3.4
rpds-py==0.17.1
rsa==4.9
scipy==1.11.4
scooby==0.9.2
six==1.16.0
stack-data==0.6.3
tenacity==8.2.3
tomli==2.0.1
traitlets==5.14.1
typing_extensions==4.9.0
uritemplate==4.1.1
urllib3==2.1.0
vtk==9.3.0
wcwidth==0.2.13
Werkzeug==3.0.1
widgetsnbextension==4.0.9
zipp==3.17.0
```

RobPasMue, 2024-01-23:
@b-matteo @umutsoysalansys could you help here?

RobPasMue, 2024-01-23:
Can you show the entire stackcall @Alberto-DM?

Alberto-DM, 2024-01-23:
> Can you show the entire call stack @Alberto-DM?

Do you mean this?

```
Connected to pydev debugger (build 233.13135.95)
Traceback (most recent call last):
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\errors.py", line 99, in wrapper
    out = func(*args, **kwargs)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\modeler.py", line 295, in open_file
    return self.read_existing_design()
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\modeler.py", line 180, in read_existing_design
    design = Design("", self, read_existing_design=True)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\errors.py", line 99, in wrapper
    out = func(*args, **kwargs)
  File "<@beartype(ansys.geometry.core.designer.design.Design.__init__) at 0x16d7ab16cb0>", line 67, in __init__
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\designer\design.py", line 138, in __init__
    self.__read_existing_design()
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\ansys\geometry\core\designer\design.py", line 693, in __read_existing_design
    Quantity(property.value, property.units),
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\plain\quantity.py", line 218, in __new__
    units = inst._REGISTRY.parse_units(units)._units
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\plain\registry.py", line 1189, in parse_units
    self.parse_units_as_container(input_string, as_delta, case_sensitive)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\nonmultiplicative\registry.py", line 70, in parse_units_as_container
    return super().parse_units_as_container(input_string, as_delta, case_sensitive)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\plain\registry.py", line 1204, in parse_units_as_container
    return self._parse_units_as_container(input_string, as_delta, case_sensitive)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\plain\registry.py", line 1239, in _parse_units_as_container
    cname = self.get_name(name, case_sensitive=case_sensitive)
  File "C:\AnsysDev\venvs\Geometry_Simplification-venv\lib\site-packages\pint\facets\plain\registry.py", line 647, in get_name
    raise UndefinedUnitError(name_or_alias)
pint.errors.UndefinedUnitError: 'J_Kg_Kdeg' is not defined in the unit registry
python-BaseException

Process finished with exit code -1
```

RobPasMue, 2024-01-23:
Precisely, thanks @Alberto-DM - I think I am inferring the issue... seems to be an error in the way we parse units. Let me try and debug later this day.

b-matteo, 2024-01-23:
@RobPasMue i can take it if you want to

RobPasMue, 2024-01-23:
On it, no worries - issue is related to unsupported materials. I am adding partial support for it

closed 2024-01-23 by RobPasMue