ansys/pyansys-geometry #1012: Bug located in download method with Discovery backend

kmahajan-cadfem, 2024-02-23:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

while creating a design with the `Discovery` backend, there is a bug in download method.
as soon as `download` is issued
- the discovery would ask for the file path where  it should save the design (this should also be suppressed in my opinion)
- regardless if the file is saved or not following error message is shown

```shell
GeometryExitedError                       Traceback (most recent call last)
Cell In[9], line 1
----> 1 design.download("test.scdocx")

File <venv>\lib\site-packages\ansys\geometry\core\errors.py:101, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     99     out = func(*args, **kwargs)
    100 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
--> 101     raise GeometryExitedError(
    102         f"Geometry service connection terminated: {error.details()}"
    103     ) from None
    105 if threading.current_thread().__class__.__name__ == "_MainThread":
    106     received_interrupt = bool(SIGINT_TRACKER)

GeometryExitedError: Geometry service connection terminated: Could not find file 'C:\Users\<username>\AppData\Local\Temp\c7ae262f-bd26-4897-bbc5-e2a44a053f72.scdocx'.
```

**The Discovery is still open and I do see above in the UI  - API Server tab which looks like following**

[image]

### 📝 Steps to reproduce

create a design using following code

```python
from ansys.geometry.core import Modeler
from ansys.geometry.core.connection.backend import BackendType
from ansys.geometry.core.errors import GeometryRuntimeError
from ansys.geometry.core.math.point import Point2D
from ansys.geometry.core.sketch import Sketch
from ansys.geometry.core import launch_modeler_with_discovery
modeler_discovery = launch_modeler_with_discovery()

modeler = modeler_discovery

design = modeler.create_design("Integrated_Example")
design.extrude_sketch("Box", Sketch().box(Point2D([0, 0]), 1, 1), 1)
values,design = modeler.run_discovery_script_file(
    "integrated_script.py", {"radius": "1"},True
)
design.download("test.scdocx")
```

content of the `integrated_script.py` is as following

```python
doc = GetActiveWindow().Document
radius = MM(int(argsDict["radius"]))

sphere = Sphere.Create(Frame.World, radius)
u_range = Interval.Create(0, 2 * math.pi)
v_range = Interval.Create(-math.pi / 2, math.pi / 2)
box = BoxUV.Create(u_range, v_range)
b = Body.CreateSurfaceBody(sphere, box)
db = DesignBody.Create(doc.MainPart, "sphere", b)

numBodies = doc.MainPart.GetAllBodies().Count

result = {"numBodies": numBodies}
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024 R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
-
```

RobPasMue, 2024-02-26:
Let me look into it @kmahajan-cadfem - thanks for opening this issue.

RobPasMue, 2024-02-26:
Bug reported to backend team - pinging @umutsoysalansys @b-matteo and @jonahrb for visibility

kmahajan-cadfem, 2024-02-26:
> I was able to reproduce your problem @kmahajan-cadfem - a quick workaround is to use the `save()` API - especially for products in which you are directly running the service rather than the Docker container.
> 
> For example, if you are using Ansys Discovery, SpaceClaim or the Ansys Geometry Service directly on your machine, it makes more sense to call the `save()` API since you do not need to download it from anywhere - it is already on your machine. When running remotely or on a Docker container, download makes sense since the file is stored somewhere remotely.
> 
> ```python
> # Instead of downloading the geometry
> # design.download("test.scdocx")
> 
> # Save the geometry locally
> import os
> 
> design.save(os.path.join(os.getcwd(), "test.scdocx"))
> ```
> 
> I hope I made myself clear - if you have doubts let me know.
> 
> Nonetheless, this is still a bug since Discovery shouldn't be asking you where to store the file. We will try to get that fixed for the next release - or this one if there is anything I can do in the Python side.

using `save` is a workaround, but I think there should be a consistent APIs regardless of the backend.
Thanks for the reply

RobPasMue, 2024-02-26:
I completely agree @kmahajan-cadfem - we will try to solve this issue asap like I mentioned.

b-matteo, 2024-03-27:
Hi @kmahajan-cadfem, @RobPasMue this issue has been solved, the fix will be available in Ansys Discovery 24.2.
Closing this issue. Thanks for letting us know and for your patience.

Regards

Matteo

closed 2024-03-27 by b-matteo