ansys/pyansys-geometry #1022: Problem located in launch_modeler_with_geometry_service

kmahajan-cadfem, 2024-02-27:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

According to the documentation, if I have installed Ansys Version, 2023 R2 or later, user should be able to do the following

```python
from ansys.geometry.core import launch_modeler_with_geometry_service

modeler = launch_modeler_with_geometry_service()
```

This does not work because there is no `GeometryService` in the unified installer.

this means user cannot actually call the headless Geometry Service with `2024 R1`, or am I missing some package that is not installed by default?

following are details of my installation
[image]

```shell
2024 R1.01

Point Releases and Patches installed:

Ansys, Inc. License Manager 2024 R1
Viewer 2024 R1
Discovery 2024 R1
SpaceClaim 2024 R1
Icepak (includes CFD-Post) 2024 R1
Fluids 2024 R1
CFD-Post only 2024 R1
CFX (includes CFD-Post) 2024 R1
Chemkin 2024 R1
EnSight 2024 R1
Fluent (includes CFD-Post) 2024 R1
Polyflow (includes CFD-Post) 2024 R1
ICEM CFD 2024 R1
TurboGrid 2024 R1
Speos 2024 R1
Speos HPC 2024 R1
optiSLang 2024 R1
Remote Solve Manager Standalone Services 2024 R1
Additive 2024 R1
Autodyn 2024 R1
Customization Files for User Programmable Features 2024 R1
LS-DYNA 2024 R1
Mechanical Products 2024 R1
Motion 2024 R1
Sherlock 2024 R1
Ansys 2024R1 Service Pack 1 2024 R1.01
Discovery 2024 R1.01
Motion 2024 R1.01
SpaceClaim 2024 R1.01
Polyflow (includes CFD-Post) 2024 R1.01
Speos 2024 R1.01
Speos HPC 2024 R1.01
```

### 📝 Steps to reproduce

- Use unified Ansys Installer for 2024 R1
- install service pack 
- run above code with virtual env

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024 R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
-
```

RobPasMue, 2024-02-27:
Hi @kmahajan-cadfem - can you verify whether you have the following installation on your machine or not?

```
C:\Program Files\ANSYS Inc\v241\GeometryService
```

If not, this means that the Ansys Geometry Service is not installed. Customers can install it from the Ansys Customer Portal by using the unified installer. That installer should provide you with it.

kmahajan-cadfem, 2024-02-27:
@RobPasMue, that folder does not exist on my machine, and using the automated installer is at the moment not an option, because the Ansys Installation is deployed by IT using deployment tools, and as far as I know this is very standard procedure for many companies to use the unified Installer for deployment.

[image]

RobPasMue, 2024-02-27:
I will have to check how it gets installed in that case using the Unified installer. But if you don't have the previously mentioned folder it means that it has not been installed.

kmahajan-cadfem, 2024-02-27:
I have double checked with IT as well as 2 different colleagues and seems like we are installing  full package and yet the GeometryService is excluded, IT is not doing anything specific to exclude this package.

could you check on your side with  Unified Installer?

kmahajan-cadfem, 2024-02-28:
@RobPasMue, as an additional information our IT uses the flags from following page to deploy Ansys 

https://ansyshelp.ansys.com/account/secured?returnurl=/Views/Secured/corp/v241/en/installation/win_silent.html

for structural packet the following flags are used

```shell
aview
discovery
spaceclaim
additive
aqwa
autodyn
ansyscust
lsdyna
matcal
mechapdl
motion
sherlock
soundsas
optislang
acad_reader
adinventor_reader
catia5_reader
proe_reader
jtopen
ug_reader
parasolid
solidedge_reader
solidworks_reader
dcs
rsm
scdm_config
```

discovery and scdm are part of this list but I could not find any reference to `GeometryService` in the help page.

**I can confirm that GeometryService is already part of the `Unified Installer`, but will not be installed if the above flags are used.**

RobPasMue, 2024-02-28:
I just remembered this discussion @kmahajan-cadfem - can you give it a try to the following flag?

https://github.com/ansys/pyansys-geometry/discussions/883#discussioncomment-8229623

```
geometryservice
```

I think this was the flag intended for the unified installer. I need to ask the team to properly document it on the Unified Installer docs.

RobPasMue, 2024-02-29:
Did you give a try to this @kmahajan-cadfem ?

kmahajan-cadfem, 2024-02-29:
I am sorry for late reply, I did forward this info to our IT and they will add this to the list of flags.
I think this can be closed here, and you can follow up with documentation team about the help page.

Thanks again for your quick responses

RobPasMue, 2024-02-29:
Awesome @kmahajan-cadfem let me close this issue.

I am planning on adding all these caveats to the [FAQ section](https://geometry.docs.pyansys.com/version/stable/getting_started/faq.html) in our docs right away

closed 2024-02-29 by RobPasMue