ansys/pyansys-geometry #1036: Scale and Mirror a body

jonahrb, 2024-03-05:
### 📝 Description of the feature

similar to how we have `body.rotate()` and `body.translate()` we should add `body.scale()` and `body.mirror()`

### 💡 Steps for implementing the feature

`body.scale()` should just take a float and will use `API.Geometry.Matrix.CreateScale()` on ApiServer

I'm assuming `body.mirror()` should take a plane and use `API.Geometry.Matrix.CreateMapping()`

jonahrb, 2024-03-05:
I considered this a while ago - the ability to allow the user to supply their own 4x4 transformation matrix, instead of performing each operation individually. That can be discussed again during the implementation of this feature request.

RobPasMue, 2024-03-06:
Hi @jonahrb   let's implement these methods! About the 4x4 by matrix, we can give it another thought. I believe that with this full set of methods, everything is available. Nonetheless, if they want to cut their way and do a full transformation that would be possible... but it might be error prone. We can discuss offline

closed 2024-03-12 by jonahrb