ansys/pyansys-geometry #1102: run_discovery_script_file does not return a dict

Alberto-DM, 2024-04-05:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

The `Modeler.run_discovery_script_file()` method should return a `dict`, but instead returns a `ScalarMapContainer` type.
See image:
[image]

Most of the time is enough to run `dict(returned_obj)` to convert it into a Python `dict`, but this conversion should be done (and verified!) inside the method.

### 📝 Steps to reproduce

- Run a script file that produces a result (that has at least a line `result={'test_key':'test_value'}` ) with the method  `result_obj=Modeler.run_discovery_script_file()` 
- Check the result object returned by the method.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Discovery 2024R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansys-api-dbu==0.2.5
ansys-api-geometry==0.3.8
ansys-geometry-core==0.4.14
ansys-pythonnet==3.1.0rc3
ansys-tools-path==0.5.1
attrs==23.2.0
beartype==0.18.2
cachetools==5.3.3
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.1
cycler==0.12.1
defusedxml==0.7.1
fonttools==4.50.0
fpdf2==2.7.8
google-api-core==2.18.0
google-api-python-client==2.125.0
google-auth==2.29.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.63.0
grpcio==1.62.1
grpcio-health-checking==1.62.1
httplib2==0.22.0
idna==3.6
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.8.4
numpy==1.26.4
packaging==24.0
pillow==10.3.0
Pint==0.23
platformdirs==4.2.0
plumbum==1.8.2
pooch==1.8.1
proto-plus==1.23.0
protobuf==4.25.3
psutil==5.9.8
pyaedt==0.8.7
pyasn1==0.6.0
pyasn1_modules==0.4.0
pycparser==2.22
pyedb==0.7.0
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pyvista==0.43.5
pywin32==306
referencing==0.34.0
requests==2.31.0
rpds-py==0.18.0
rpyc==6.0.0
rsa==4.9
scipy==1.13.0
scooby==0.9.2
six==1.16.0
typing_extensions==4.10.0
uritemplate==4.1.1
urllib3==2.2.1
vtk==9.3.0
```

RobPasMue, 2024-04-05:
Have a look at the documentation @Alberto-DM 

https://geometry.docs.pyansys.com/version/stable/api/ansys/geometry/core/modeler/Modeler.html#Modeler.run_discovery_script_file

It returns either a dictionary or a dictionary and a Design object. Depending on whether you ask for ``import_design=True`` or not.

RobPasMue, 2024-04-05:
If you still have the issue, please post the script.

Alberto-DM, 2024-04-05:
@RobPasMue I have checked the documentation and `import_design` is set to `False` (which is the default). Anyway, the documentation says that returns either a dict or a tuple. It is not returning neither of them. 

The script file:
`result={'test_key':'test_value'}`

[image]

As you can see the type is not dict, it is ScalarMapContainer

RobPasMue, 2024-04-08:
Gotcha - just tested it on my side directly. I will implement the fix right away.

Nonetheless, in the meantime, the ScalarMapContainer behaves exactly like a dictionary. So you can assume it is a dictionary in your script.

RobPasMue, 2024-04-08:
Fixed in #1103 - the next release will contain the fix =) Thanks for opening the issue @Alberto-DM!

PR #1103 (2024-04-08): fix: return type should be dict and not ``ScalarMapContainer`` (grpc type)
## Description
Return type coming from ``run_discovery_script_file`` was ``ScalarMapContainer`` instead of ``dict``. Though they effectively behave in the same way, it is true that the type should be a dictionary since we should avoid returning gRPC objects to the end user.

## Issue linked
Closes #1102 

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added a [... cut]
closed 2024-04-08 by RobPasMue