ansys/pyansys-geometry #1110: Geometry Visualization Issue

abhishekchitwar, 2024-04-12:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

Couldn't able to visualize the geometry, standalone trame, PyVista trame or provided TrameVisualizer Class, 

``` python

from ansys.geometry.core.plotting.trame_gui import TrameVisualizer
view = TrameVisualizer()
view.set_scene(plotter=plotter)
view.show()

```

```
CRITICAL -  -  logger - handle_exception - Uncaught exception
Traceback (most recent call last):
  File "d:\Research\pygeom_plotter_test\geom_plotter.py", line 86, in <module>
    view.set_scene(plotter=plotter)
  File "C:\Users\username\.ansys_python_venvs\.venv_pyansys\lib\site-packages\ansys\geometry\core\plotting\trame_gui.py", line 64, in set_scene
    view = plotter_ui(plotter.scene)
AttributeError: 'PlotterHelper' object has no attribute 'scene' 
```

### 📝 Steps to reproduce

Pass the plotter object to respective implementation, e.g., 

```
TrameVisualizer.set_scene(plotter)
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys 24R1 

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
Installed all the pyansys meta-pacakages using Ansys Python Manager.

Additionally install following packages.
trame trame-vuetify trame-vtk
```

RobPasMue, 2024-04-12:
Hi @abhishekchitwar - the way to use the trame visualizer is not this one.

Please do as follows

```py

import ansys.geometry.core as pygeo

pygeo.USE_TRAME = True

# Do your stuff...
#
#

design.plot()
```

Our PyAnsys Geometry main objects (design, component, body) all have a "plot()" method you can use easily this way with your intended backend

RobPasMue, 2024-04-12:
Have a look at https://geometry.docs.pyansys.com/version/stable/examples/01_getting_started/05_plotter_picker.html as well

abhishekchitwar, 2024-04-15:
Hi @RobPasMue 

I attempted the following code snippet, but the design plot still appears in a desktop window. I'm looking for a way to display this design on a server or local URL instead.

``` 
from ansys.geometry.core import launch_modeler
from ansys.geometry.core.misc import UNITS, Distance, DEFAULT_UNITS
from ansys.geometry.core.math import Point2D
from ansys.geometry.core.sketch import Sketch
from ansys.geometry.core.plotting import PlotterHelper
import pyvista as pv
#from trame_app import TrameApp
DEFAULT_UNITS.LENGTH = UNITS.mm
modeler = launch_modeler(mode='spaceclaim')

launch_modeler.USE_TRAME=True

base_sketch = Sketch()
base_sketch.segment(Point2D([0,0]), Point2D([0,10]))
base_sketch.segment(Point2D([0,10]), Point2D([10,10]))
base_sketch.segment(Point2D([10,10]), Point2D([10,0]))
base_sketch.segment(Point2D([10,0]), Point2D([0,0]))

# Start by creating the Design
design = modeler.create_design("ModelingDemo")

# Create a body directly on the design by extruding the sketch
body = design.extrude_sketch(
    name="Design_Body", sketch=base_sketch, distance=Distance(10, unit=UNITS.mm)
)

# design plot
design.plot()

# plotter, this is for test
# plot_list = [body]
# plotter = PlotterHelper(use_trame=True)
# plotter.plot(plot_list)

```

abhishekchitwar, 2024-04-15:
Hi @RobPasMue,

I just updated the argument with `design.plot(use_trame= True)`, getting following issue.

``` 
    raise TypeError(
TypeError: Server using client_type='vue3' while we expect 'vue2' 

```
It seems, it requires update from TrameVisualizer.

abhishekchitwar, 2024-04-15:
Hi @RobPasMue,

I just updated the `self.server = get_server(client_type='vue2')` with TrameVisualizer, it worked as updated. Thanks!

Thanks!

RobPasMue, 2024-04-15:
Oh interesting! So you needed to pass the client type to the TrameVisualizer.. We might have to expose this somehow. Could you share your final script? I guess you had to modify your local version of PyAnsys Geometry, right?

abhishekchitwar, 2024-04-15:
Hi @RobPasMue,

yeah, I modified the local pygeometry installation.

It would be great if TrameVisualizer exposes several methods in context of remote visualization, e.g.

``` 
    def get_url(self):
        return f"http://{self.address}:{self.port}"
```
Ease for developer to configure address and port as per requirement.

Here is the final code! Example test case developed by @agvarghe 

``` python
from ansys.geometry.core import launch_modeler
from ansys.geometry.core.misc import UNITS, Distance, DEFAULT_UNITS
from ansys.geometry.core.math import Point2D
from ansys.geometry.core.sketch import Sketch
from ansys.geometry.core.plotting import PlotterHelper
import pyvista as pv

DEFAULT_UNITS.LENGTH = UNITS.mm
modeler = launch_modeler(mode='spaceclaim')

# launch_modeler.USE_TRAME=True

base_sketch = Sketch()
base_sketch.segment(Point2D([0,0]), Point2D([0,10]))
base_sketch.segment(Point2D([0,10]), Point2D([10,10]))
base_sketch.segment(Point2D([10,10]), Point2D([10,0]))
base_sketch.segment(Point2D([10,0]), Point2D([0,0]))

# Start by creating the Design
design = modeler.create_design("ModelingDemo")

# Create a body directly on the design by extruding the sketch
body = design.extrude_sketch(
    name="Design_Body", sketch=base_sketch, distance=Distance(10, unit=UNITS.mm)
)

# design plot
design.plot(use_trame=True)

```
Thanks!

RobPasMue, 2024-04-19:
We will delegate the actual remote visualization to a different project we are working on. On this one we will just fix the inputs needed to use vue3 or vue2 =)

This issue will be closed by #1148 - thanks @AlejandroFernandezLuces

closed 2024-04-19 by RobPasMue