ansys/pyansys-geometry #1184: design.create_sphere

rmanno91, 2024-04-30:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

If I try to use the create_sphere method directly on the design object after having created a box body I get this exception:

[image]

The exception is not raised if I create the sphere first and the box after.

### 📝 Steps to reproduce

````python
from ansys.geometry.core.math import Point3D
from ansys.geometry.core.misc import Distance
from ansys.geometry.core.math import (
    UNITVECTOR3D_X,
    UNITVECTOR3D_Y,
    Plane,
    Point2D,
    Point3D,
)
from ansys.geometry.core.misc import DEFAULT_UNITS
from ansys.geometry.core.sketch import Sketch
from ansys.geometry.core.connection.launcher import launch_modeler_with_geometry_service

modeler = launch_modeler_with_geometry_service()
design = modeler.create_design("RVE")

unit = DEFAULT_UNITS.LENGTH

plane = Plane(
    Point3D([1 / 2, 1 / 2, 0.0], unit=unit),
    UNITVECTOR3D_X,
    UNITVECTOR3D_Y,
)
box_plane = Sketch(plane)
box_plane.box(Point2D([0.0, 0.0]), width=1 * unit, height=1 * unit)

box = design.extrude_sketch("Matrix", box_plane, 1 * unit)
sphere_body = design.create_sphere("particle", Point3D([0.0,0.0,0.0], unit=unit), Distance(0.5))
design.plot(show_edges=True)
````

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

24r2

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
accessible-pygments==0.0.4
alabaster==0.7.16
annotated-types==0.6.0
ansys-api-dbu==0.3.0
ansys-api-geometry==0.4.1
ansys-api-mapdl==0.5.1
ansys-api-meshing-prime==0.1.2
ansys-api-platform-instancemanagement==1.0.0
ansys-geometry-core @ git+https://github.com/ansys/pyansys-geometry@0ff8a6d869526d4c01cfc06b894aeb933fd533a4
ansys-mapdl-core==0.67.0
ansys-mapdl-reader==0.53.0
-e git+https://github.com/ansys-internal/pymaterialdesigner@5385d7cfd48e881f16f3d66e217ad4fb92abb27a#egg=ansys_material_designer_core
ansys-materials-manager @ git+https://github.com/ansys/pymaterials-manager@a2560dc48e2c53686d20ae53ba642021186dc3c1
ansys-math-core==0.1.5
ansys-meshing-prime @ git+https://github.com/ansys/pyprimemesh@b77076cbf11ee4eec404f929c4d10a758ecc3a7e
ansys-platform-instancemanagement==1.1.2
ansys-sphinx-theme==0.15.2
ansys-tools-path==0.5.2
appdirs==1.4.4
Babel==2.14.0
beartype==0.18.5
beautifulsoup4==4.12.3
build==1.2.1
cachetools==5.3.3
certifi==2024.2.2
cfgv==3.4.0
charset-normalizer==3.3.2
click==8.1.7
colorama==0.4.6
contourpy==1.2.1
coverage==7.4.4
cycler==0.12.1
distlib==0.3.8
docutils==0.21.1
filelock==3.13.4
flit==3.9.0
flit_core==3.9.0
fonttools==4.51.0
geomdl==5.3.1
google-api-core==2.18.0
google-api-python-client==2.126.0
google-auth==2.29.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.63.0
grpcio==1.62.2
grpcio-health-checking==1.48.2
httplib2==0.22.0
identify==2.5.36
idna==3.7
imagesize==1.4.1
importlib_metadata==7.1.0
iniconfig==2.0.0
jaraco.classes==3.4.0
jaraco.context==5.3.0
jaraco.functools==4.0.1
Jinja2==3.1.3
keyring==25.1.0
kiwisolver==1.4.5
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.4
mdurl==0.1.2
more-itertools==10.2.0
nh3==0.2.17
nodeenv==1.8.0
numpy==1.26.4
numpydoc==1.7.0
packaging==24.0
pillow==10.3.0
Pint==0.23
pkginfo==1.10.0
platformdirs==4.2.0
pluggy==1.5.0
pooch==1.8.1
pre-commit==3.7.0
proto-plus==1.23.0
protobuf==3.20.3
psutil==5.9.8
pyansys-tools-versioning==0.5.0
pyasn1==0.6.0
pyasn1_modules==0.4.0
pydantic==2.7.0
pydantic_core==2.18.1
pydata-sphinx-theme==0.14.4
Pygments==2.17.2
pyiges==0.3.1
pyparsing==3.1.2
pyproject_hooks==1.0.0
pytest==8.1.1
pytest-cov==5.0.0
python-dateutil==2.9.0.post0
pyvista==0.43.5
pywin32-ctypes==0.2.2
PyYAML==6.0.1
readme_renderer==43.0
requests==2.31.0
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.7.1
rsa==4.9
scipy==1.13.0
scooby==0.9.2
semver==3.0.2
setuptools==69.5.1
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.3.7
sphinx-copybutton==0.5.2
sphinxcontrib-applehelp==1.0.8
sphinxcontrib-devhelp==1.0.6
sphinxcontrib-htmlhelp==2.0.5
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.7
sphinxcontrib-serializinghtml==1.1.10
tabulate==0.9.0
tomli_w==1.0.0
tqdm==4.66.2
twine==5.0.0
typing_extensions==4.11.0
uritemplate==4.1.1
urllib3==2.2.1
virtualenv==20.25.3
vtk==9.3.0
wheel==0.43.0
zipp==3.18.1
```

RobPasMue, 2024-05-01:
Interesting! Thanks for bringing this up @rmanno91 - I'm OoO until the 6th of May. I'll investigate upon my return. Pinging @ansys/pyansys-geometry-maintainers in case you can look into it.

RobPasMue, 2024-05-07:
Hi @rmanno91 - I have been able to reproduce this same error... investigating with the team.

RobPasMue, 2024-07-02:
Created test https://github.com/ansys/pyansys-geometry/pull/1291 to track progress and resolution

umutsoysalansys, 2024-07-03:
@rmanno91, @RobPasMue This is fixed in and changes are merged in APIServer. Tests should be passing now. Let me know if you still issues after getting the latest api server.

RobPasMue, 2024-07-04:
New version of the server is not available @umutsoysalansys since 9 days ago. Please verify with @b-matteo that the server image got pushed

umutsoysalansys, 2024-07-08:
@smereu is there a manual pipeline to deliver latest server package to here? Is there anyone else other than Matteo who can push it?

RobPasMue, 2024-07-08:
It should be an automated process as far as I know... Did it get merged into the ``develop`` branch in ADO?

umutsoysalansys, 2024-07-08:
@RobPasMue yes on wednesday.

RobPasMue, 2024-07-08:
No package is being pushed to GitHub... I can confirm that

closed 2024-07-11 by RobPasMue

rmanno91, 2024-07-12:
> Issue solved in the latest server version (25.1) - @rmanno91 we included a test to confirm it's working. Give it a try whenever you have the chance

Thanks @umutsoysalansys and @RobPasMue, I will have a look soon and let you know :)

rmanno91, 2024-09-05:
@RobPasMue @umutsoysalansys It was not that soon, it seems to work fine, thanks for that!

PR #1291 (2024-07-11): test: create sphere bug raised after box creation
## Description
Sphere creation after box creation fails

## Issue linked
Closes #1184

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [X] I have assigned this PR to myself.
- [X] I have added the minimum versio [... cut]