ansys/pyansys-geometry #1229: Create a conda recipe to publish to conda-forge

RobPasMue, 2024-05-30:
This will make ansys-geometry-core available on conda-forge! :)

RobPasMue, 2024-05-31:
Implied several packages:

- [X] ``ansys_tools_protoc_helper``
- [X] ``ansys-tools-path``
- [ ] ``ansys-api-dbu``
- [ ] ``ansys-api-geometry``

And finally...

- [ ] ``ansys-geometry-core``

RobPasMue, 2024-05-31:
Current feedstocks on conda:

https://github.com/conda-forge/ansys-tools-path-feedstock
https://github.com/conda-forge/ansys_tools_protoc_helper-feedstock

Ongoing PR for other repos:
https://github.com/conda-forge/staged-recipes/pull/26523

RobPasMue, 2024-06-03:
Well, we are missing two other packages now:

- [ ] ansys-api-platform-instancemanagement
- [ ] ansys-platform-instancemanagement

RobPasMue, 2024-06-07:
Well... last one to go https://github.com/conda-forge/staged-recipes/pull/26585

MaxJPRey, 2024-06-09:
It seems you are done 😃 . Nice.

RobPasMue, 2024-06-10:
https://github.com/conda-forge/ansys-geometry-core-feedstock

Yep, yesterday it went through!

PR #1247 (2024-06-10): docs: add conda information for package
## Description
Once published on conda-forge (see https://github.com/conda-forge/staged-recipes/pull/26585) we will be able to refer to it.

## Issue linked
Closes #1229 

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [ [... cut]