ansys/pyansys-geometry #1295: Export to .pmdat fails

greschd, 2024-07-04:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

When loading the attached STEP file [ply_boundaries.step.txt](https://github.com/user-attachments/files/16099189/ply_boundaries.step.txt), calling the `export_to_pmdb` method causes the following exception:

```
CRITICAL -  -  logger - handle_exception - Uncaught exception
Traceback (most recent call last):
  File "<script_location>\pmdat_export_error.py", line 5, in <module>
    design.export_to_pmdb(".")
  File "<venv_location>\lib\site-packages\ansys\geometry\core\designer\design.py", line 474, in export_to_pmdb
    self.download(file_location, DesignFileFormat.PMDB)
  File "<venv_location>\lib\site-packages\ansys\geometry\core\errors.py", line 98, in wrapper
    raise GeometryExitedError(
ansys.geometry.core.errors.GeometryExitedError: Geometry service connection terminated: Could not find file '<TEMPDIR>\2a350bea-f124-4461-97c6-2aeb0781eb75.pmdb'.
``

[1] renamed to `.txt` to make Github happy

### 📝 Steps to reproduce

- Download the STEP file linked above
- Run the following code, changing the path to the file as needed:
  ```python
  from ansys.geometry.core import launch_modeler_with_geometry_service

  modeler = launch_modeler_with_geometry_service()
  design = modeler.open_file("ply_boundaries.step")
  design.export_to_pmdb(".")
  ```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2025R1

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
aiohttp==3.9.5
aiosignal==1.3.1
ansys-api-dbu==0.3.3
ansys-api-geometry==0.4.4
ansys-geometry-core==0.6.5
ansys-tools-path==0.6.0
ansys-tools-visualization-interface==0.2.6
appdirs==1.4.4
async-timeout==4.0.3
attrs==23.2.0
beartype==0.18.5
certifi==2024.7.4
charset-normalizer==3.3.2
click==8.1.7
colorama==0.4.6
contourpy==1.2.1
cycler==0.12.1
flexcache==0.3
flexparser==0.3.1
fonttools==4.53.0
frozenlist==1.4.1
grpcio==1.64.1
grpcio-health-checking==1.64.1
idna==3.7
kiwisolver==1.4.5
matplotlib==3.9.1
more-itertools==10.3.0
msgpack==1.0.8
multidict==6.0.5
numpy==2.0.0
packaging==24.1
pillow==10.4.0
Pint==0.24.1
platformdirs==4.2.2
pooch==1.8.2
protobuf==5.27.2
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pyvista==0.43.10
requests==2.32.3
scipy==1.14.0
scooby==0.10.0
semver==3.0.2
six==1.16.0
trame==3.6.3
trame-client==3.2.1
trame-server==3.0.3
trame-vtk==2.8.9
trame-vuetify==2.6.1
typing_extensions==4.12.2
urllib3==2.2.2
vtk==9.3.1
websockets==12.0
wslink==2.1.1
yarl==1.9.4
```

RobPasMue, 2024-07-04:
Hmm seems like you are using the local version of the Geometry Service (meaning, it is on your filesystem). Could you try running the script with the following modification?

```py
design = modeler.open_file("ply_boundaries.step", upload_file=False)
design.export_to_pmdb()
```

RobPasMue, 2024-07-04:
Nonetheless it seems like the export/download API is the one failing... Let's give it a try to 

https://geometry.docs.pyansys.com/version/0.6/api/ansys/geometry/core/designer/design/Design.html#Design.save

try calling

```py
design.save("ply_boundaries.pmdb")
```

Fingers crossed...

greschd, 2024-07-04:
> upload_to_server=False

Produces the same error

> design.save("ply_boundaries.pmdb")

Produces the following error:

```
ansys.geometry.core.errors.GeometryExitedError: Geometry service connection terminated: Path cannot be the empty string or all whitespace.
```

Can you reproduce the issue with the STEP above? Maybe I just got a bad daily build. 

Otherwise, I suspect it's specific to this file. With another STEP file the original code works, but the `design.save` variant produces the same exception.

RobPasMue, 2024-08-30:
Hi @greschd - I came back to this issue after some time (sorry for the delay). I am able to reproduce the same problem you report. However, looking into the server logs I found an interesting error...

```sh
2024-08-30 13:06:17 1:06:17 PM: start save C:\Users\ContainerAdministrator\AppData\Local\Temp\43d7895b-f783-46d7-85d2-f7becf54befa.scdocx
2024-08-30 13:06:17 1:06:17 PM: ended save C:\Users\ContainerAdministrator\AppData\Local\Temp\43d7895b-f783-46d7-85d2-f7becf54befa.scdocx
2024-08-30 13:06:18 Failed to import assembly from SpaceClaim.    at AnsysSCPlugin.Transfer.AssemblyTransferer.Transfer(Part rootPart, TransferComponentCollection componentCollection) in D:\AnsysDev\NoBackup\SpaceClaim\Partners\ANSYS\25.1\AnsysSCPlugin.25.1\Transfer\AssemblyTransferer.cs:line 96
2024-08-30 13:06:18    at AnsysSCPlugin.TransferManager.<TransferAssemblyComponents>b__41_0() in D:\AnsysDev\NoBackup\SpaceClaim\Partners\ANSYS\25.1\AnsysSCPlugin.25.1\TransferManager.cs:line 418
2024-08-30 13:06:18    at SpaceClaim.Api.V251.WriteBlockStatic.ExecuteClientCode(Task task)
2024-08-30 13:06:18    at SpaceClaim.Api.V251.WriteBlockStatic.ExecuteTask(String commandText, TaskShim shim, Boolean addToLastCommand)
2024-08-30 13:06:18    at SpaceClaim.Api.V251.WriteBlock.ExecuteTask(String text, Task task)
2024-08-30 13:06:18    at AnsysSCPlugin.TransferManager.TransferAssemblyComponents() in D:\AnsysDev\NoBackup\SpaceClaim\Partners\ANSYS\25.1\AnsysSCPlugin.25.1\TransferManager.cs:line 405
```

My guess is that something is going on when exporting to PMDB only - because prior to these errors you can see that the export to scdocx worked just fine. I'll report it to the PMDB plugin team so that they can investigate.

RobPasMue, 2024-08-30:
FYI - I also tried it with SpaceClaim and it also failed so it might be something broader (not specific to the Geometry Service)...

greschd, 2025-02-25:
> For PMDB export, those "Curves" would have to be converted to "Beams" to be valid.

In that case, I guess the error handling needs to be improved s.t. it reports the error instead of crashing?