ansys/pyansys-geometry #1303: Enhance Plotting Functionality to Differentiate Bodies by Color

agvarghe, 2024-07-15:
### 📝 Description of the feature

capability to display different bodies with assorted colors when plotting a component.

RobPasMue, 2024-07-16:
That is implemented in fact - but you would have to make use of the GeometryPlotter object directly. You mean doing this directly when requesting to plot a ``design`` or ``component`` right?

RobPasMue, 2024-07-17:
Okay so we need to implement the color cycler shown here https://github.com/ansys/pyansys-geometry/issues/368

Shouldn't be hard honestly. The color cycler should be an optional flag that a user can decide whether to activate or not when requesting ``design.plot()``, ``component.plot()``, etc.

RobPasMue, 2024-07-31:
Transferring to ansys-tools-visualization-interface since this functionality will be exposed there. I will add an example on this repo to show how to do it.