ansys/pyansys-geometry #1328: add Parameters to Tesselate() method

StefanThoene, 2024-07-23:
### 📝 Description of the feature

component and Body do have a tesselate method.
as a user i would like to be able to define the Tesselation Parameters to be able to extract all informations
This is important to use Pygeometry in workflows which need surface mesh which also need surface normals

### 💡 Steps for implementing the feature

add optional properties to the Tesselate method:
[image]

RobPasMue, 2024-07-29:
This would have to be implemented server side in order to expose all these options. Pinging @b-matteo @umutsoysalansys @smereu @jmabshof for visibility

RobPasMue, 2025-03-19:
This has been implemented for 25R2 in #1778 if I am not mistaken, correct @mlkaplan36 ?

StefanThoene, 2025-03-19:
can you confirm that would be awesome to have

RobPasMue, 2025-03-19:
I have the feeling this was missing: https://github.com/ansys/pyansys-geometry/pull/1850

After this is merged, the tessellation options will be generally available.

RobPasMue, 2025-03-19:
Starting on 25R2 of course.

RobPasMue, 2025-03-19:
Closing since handled and confirmed. Will be available in next release.

closed 2025-03-19 by RobPasMue