ansys/pyansys-geometry #1329: Plot/tesselate not working on geometry

StefanThoene, 2024-07-25:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

If i try the plot or tessleta method on the design it is not working

### 📝 Steps to reproduce

[LG_prep_v2.zip](https://github.com/user-attachments/files/16380144/LG_prep_v2.zip)
download data
try plot you can use following code please aduhsut the path to the file

```
from ansys.geometry.core import launch_modeler_with_geometry_service
modeler = launch_modeler_with_geometry_service()
print(modeler)
design = modeler.open_file(r"D:\temp\LG_prep_v2.scdocx")
design.tessellate()
design.plot()
```

i get then the following error:

```
CRITICAL -  -  logger - handle_exception - Uncaught exception
Traceback (most recent call last):
  File "C:\Users\sthoene\AppData\Roaming\JetBrains\PyCharmCE2023.2\scratches\Lightguide_pygeo.py", line 26, in <module>
    design.plot()
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\designer\component.py", line 1466, in plot
    custom_object=self, mesh=self.tessellate(merge_component, merge_bodies)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\designer\component.py", line 1376, in tessellate
    datasets = [body.tessellate(merge_bodies) for body in self.bodies]
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\designer\component.py", line 1376, in <listcomp>
    datasets = [body.tessellate(merge_bodies) for body in self.bodies]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\misc\checks.py", line 81, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\designer\body.py", line 1317, in tessellate
    return self._template.tessellate(merge, self.parent_component.get_world_transform())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Work\Gitdir\pyspeos\.venv\Lib\site-packages\ansys\geometry\core\errors.py", line 101, in wrapper
    raise GeometryExitedError(
ansys.geometry.core.errors.GeometryExitedError: Geometry service connection terminated: The given key was not present in the dictionary.
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

i am using 24R2

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
aiohttp==3.9.5
aiosignal==1.3.1
ansys-api-dbu==0.3.3
ansys-api-geometry==0.4.4
ansys-api-meshing-prime==0.1.2
ansys-geometry-core==0.6.3
ansys-meshing-prime==0.6.0
ansys-tools-path==0.6.0
ansys-tools-visualization-interface==0.2.6
anyio==4.4.0
appdirs==1.4.4
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.15.0
beartype==0.18.5
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2024.6.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
colorama==0.4.6
comm==0.2.2
comtypes==1.4.5
contourpy==1.2.1
cycler==0.12.1
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
executing==2.0.1
fastjsonschema==2.20.0
flexcache==0.3
flexparser==0.3.1
fonttools==4.53.0
fqdn==1.5.1
frozenlist==1.4.1
future==1.0.0
grpcio==1.65.1
grpcio-health-checking==1.48.2
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.7
iniconfig==2.0.0
ipykernel==6.29.4
ipython==8.25.0
ipywidgets==8.1.3
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.22.0
jsonschema-specifications==2023.12.1
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.1
jupyter_server_proxy==4.2.0
jupyter_server_terminals==0.5.3
jupyterlab==4.2.2
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.2
jupyterlab_widgets==3.0.11
kiwisolver==1.4.5
MarkupSafe==2.1.5
matplotlib==3.9.0
matplotlib-inline==0.1.7
mistune==3.0.2
more-itertools==10.3.0
msgpack==1.0.8
multidict==6.0.5
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
notebook==7.2.1
notebook_shim==0.2.4
numpy==2.0.1
overrides==7.7.0
packaging==24.1
pandocfilters==1.5.1
parso==0.8.4
pillow==10.3.0
Pint==0.24.1
platformdirs==4.2.2
pluggy==1.5.0
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
protobuf==3.20.3
psutil==6.0.0
pure-eval==0.2.2
PyContracts==1.8.12
pycparser==2.22
PyGeometry==1.5.6
Pygments==2.18.0
pyparsing==3.1.2
pytest==8.3.2
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pyvista==0.44.1
pywin32==306
pywinpty==2.0.13
PyYAML==6.0.1
pyzmq==26.0.3
qtconsole==5.5.2
QtPy==2.4.1
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.18.1
scipy==1.13.1
scooby==0.10.0
semver==3.0.2
Send2Trash==1.8.3
simpervisor==1.0.0
six==1.16.0
sniffio==1.3.1
soupsieve==2.5
stack-data==0.6.3
terminado==0.18.1
tinycss2==1.3.0
tornado==6.4.1
traitlets==5.14.3
trame==3.6.3
trame-client==3.1.0
trame-server==3.0.1
trame-vtk==2.8.9
trame-vuetify==2.6.0
types-python-dateutil==2.9.0.20240316
typing_extensions==4.12.2
uri-template==1.3.0
urllib3==2.2.2
vtk==9.3.0
wcwidth==0.2.13
webcolors==24.6.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==12.0
widgetsnbextension==4.0.11
wslink==2.0.5
yarl==1.9.4
```

RobPasMue, 2024-07-29:
Let me look into this to see what's happening

RobPasMue, 2024-07-29:
Seems to be specific to this ``scdocx`` file - more precisely to the ``Light Guide.1`` body, which is the one causing the tessellation to fail. Faces might be either too small or too many, and this is causing the tessellation to fail. This error occurs on all backends. @b-matteo investigating. Might be solved via #1328

RobPasMue, 2025-04-14:
Tesselation with options will be available starting on 25R2. The API is already available though @StefanThoene -- feel free to give it a try if you want :)

RobPasMue, 2025-04-14:
```py
from ansys.geometry.core.misc.options import TessellationOptions

options = TessellationOptions(...)

tess = body.tessellate(.., tess_options=options)
```

I still think this would have to be exposed more easily on the plot method though

closed 2026-01-15 by RobPasMue