ansys/pyansys-geometry #1330: Ansys Discovery 24R2 not opening file ``.dsco``

RobPasMue, 2024-07-29:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

```py
d=modeler.open_file(r"C:\Users\lvecchie\OneDrive - ANSYS, Inc\Ansys\Presales\Schneider\115-1_setbt_g1353002_asm.dsco")
```

This is not working according to @lorenzovecchietti- File has been shared offline but apparently no file (no matter what the content) is being open.

### 📝 Steps to reproduce

-

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

24R2 Discovery

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
-
```

RobPasMue, 2024-07-29:
Seems to be a bug in the Discovery server side... @b-matteo looking into it. Minimum script that replicates behavior

```py
import os
from ansys.geometry.core import launch_modeler_with_discovery

modeler = launch_modeler_with_discovery()

design = modeler.open_file(f"{os.getcwd()}\\file.dsco")

print(design)
design.plot()
modeler.close()
```

RobPasMue, 2024-07-29:
Failure happening at modeler.open_file(). File is imported correctly into Discovery but when attempting to read the existing assembly into the Python client, failure occurs.

lorenzovecchietti, 2024-07-31:
I'm not sure if this is something related just to me, but I'm reproducing the same exact behavior also with Discovery 2024R1 (maybe I need to remove 24R2 and do a clean installation?)

RobPasMue, 2024-08-29:
Problems occur on the GetAssembly rpc call - which is the one syncing the model between Ansys Discovery and the PyAnsys Geometry session. Team is investigating

RobPasMue, 2024-10-01:
This problem was fixed in 25R1 but will not be fixed on 24R2. Closing.

closed 2024-10-01 by RobPasMue