ansys/pyansys-geometry #1350: Example proposal: For Solder ball modeling

agvarghe, 2024-08-09:
### 📝 Description of the example

A script to create solderball model and also target solder for performing morphing

### 📁 Files needed for running the example

[PyGeometry.zip](https://github.com/user-attachments/files/16554617/PyGeometry.zip)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024r2

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
ansys=geometry-core==0.6.6
```

RobPasMue, 2024-10-17:
Hi @agvarghe - what's the status here? Do you need any help?

agvarghe, 2024-10-21:
Hi @RobPasMue 
I had to lower the priorty in the last month, i plan to complete this in this week

RobPasMue, 2024-10-21:
Awesome, thanks @agvarghe!

jmabshof, 2025-02-05:
@agvarghe was this example created?