ansys/pyansys-geometry #1388: No error message, if the file format is not supported

dkowollik, 2024-09-02:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

There is no error message, that the file format is not supported by the GeometryService. 

### 📝 Steps to reproduce

```python
In [1]: from ansys.geometry.core import launch_modeler_with_discovery
In [2]: modeler = launch_modeler_with_discovery(hidden=True)
In [3]: model = modeler.open_file('any_existing_file.txt')
In [4]: print(model)
ansys.geometry.core.designer.Design 0x24bc89985b0
  Name                 : Design1
  Is active?           : True
  N Bodies             : 0
  N Components         : 0
  N Coordinate Systems : 0
  N Named Selections   : 0
  N Materials          : 0
  N Beam Profiles      : 0
  N Design Points      : 0
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys 2024 R2

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
aiohappyeyeballs==2.4.0
aiohttp==3.10.5
aiosignal==1.3.1
ansys-api-dbu==0.3.4
ansys-api-geometry==0.4.7
ansys-api-mapdl==0.5.2
ansys-api-mechanical==0.1.2
ansys-api-meshing-prime==0.1.3
ansys-api-platform-instancemanagement==1.1.0
ansys-dpf-composites==0.6.0
ansys-dpf-core==0.13.0
ansys-geometry-core==0.7.0
ansys-mapdl-core==0.68.4
ansys-mapdl-reader==0.54.0
ansys-math-core==0.1.5
ansys-mechanical-core==0.11.5
ansys-mechanical-env==0.1.7
ansys-meshing-prime==0.6.2
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc3
ansys-tools-path==0.6.0
ansys-tools-visualization-interface==0.4.0
appdirs==1.4.4
asttokens==2.4.1
async-timeout==4.0.3
attrs==24.2.0
beartype==0.18.5
cachetools==5.5.0
certifi==2024.7.4
cffi==1.17.0
charset-normalizer==3.3.2
click==8.1.7
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
cycler==0.12.1
decorator==5.1.1
exceptiongroup==1.2.2
executing==2.0.1
flexcache==0.3
flexparser==0.3.1
fonttools==4.53.1
frozenlist==1.4.1
geomdl==5.3.1
google-api-core==2.19.2
google-api-python-client==2.142.0
google-auth==2.34.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.65.0
grpcio==1.66.0
grpcio-health-checking==1.62.3
httplib2==0.22.0
idna==3.8
importlib_metadata==8.4.0
ipython==8.26.0
jedi==0.19.1
kiwisolver==1.4.5
matplotlib==3.9.2
matplotlib-inline==0.1.7
more-itertools==10.4.0
msgpack==1.0.8
multidict==6.0.5
numpy==1.26.4
packaging==24.1
parso==0.8.4
pillow==10.4.0
Pint==0.24.3
platformdirs==4.2.2
pooch==1.8.2
prompt_toolkit==3.0.47
proto-plus==1.24.0
protobuf==4.25.4
psutil==6.0.0
pure_eval==0.2.3
pyansys-tools-versioning==0.6.0
pyasn1==0.6.0
pyasn1_modules==0.4.0
pycparser==2.22
Pygments==2.18.0
pyiges==0.3.1
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pyvista==0.43.10
requests==2.32.3
rsa==4.9
scipy==1.14.1
scooby==0.10.0
semver==3.0.2
six==1.16.0
stack-data==0.6.3
tabulate==0.9.0
tqdm==4.66.5
traitlets==5.14.3
trame==3.6.3
trame-client==3.2.5
trame-server==3.1.0
trame-vtk==2.8.10
trame-vuetify==2.6.2
typing_extensions==4.12.2
uritemplate==4.1.1
urllib3==2.2.2
vtk==9.3.1
wcwidth==0.2.13
websockets==12.0
wslink==2.1.2
yarl==1.9.4
zipp==3.20.1
```

RobPasMue, 2024-09-02:
Hi @dkowollik! Could you share as well the server logs?

dkowollik, 2024-09-02:
Hi @RobPasMue,

My mistake. I used the Geometry Service. I updated the python snippets above. 
The log file shows nothing special, see below (I did not copy the entire path variable from the log file):

Licensing: ANSYS242_DIR already set to 'C:\Program Files\ANSYS Inc\v242\ANSYS'
Licensing: Server.SetLicenseServer() has not been called.
License checked out in 1516 seconds
Modeler = Parasolid
ANSYS Stride 1.17.2.9183872
ModelerStarter Hoops 3: hoopsFolder=C:\Program Files\ANSYS Inc\v242\GeometryService\Stride\Hoops\
HOOPS Version: 24.1.0
HOOPS Folder: C:\Program Files\ANSYS Inc\v242\GeometryService\Stride\Hoops\
Translator directory is set to:C:\Program Files\ANSYS Inc\v242\GeometryService\Translators
Parasolid version 36.0.198
Environment variable: PATH = C:\Program Files\ANSYS Inc\v242\GeometryService\Stride;...
gRPC services subscription done.
Server started. Listening on endpoint: localhost:56913

RobPasMue, 2024-09-02:
Okay so the logs are not saying anything about the invalid file format. The service should definitely be checking it rather than the client. I will bring this to the dev team in charge of the Geometry Service. Thanks a lot for your inputs!

A quick workaround would be to implement the checks on the Python side but scaling would make it harder...

RobPasMue, 2025-09-09:
@pyansys-automation never close issue