ansys/pyansys-geometry #1402: Add pretty print component tree

dkowollik, 2024-09-04:
### 📝 Description of the feature

Is it possible to create a pretty print of the component tree with different options like show only names, include entities, display all levels? For complex large models this would be helpful. I think of something similar like a pretty print dump you can do on json strings. An alternative could be an identical unfolded SpaceClaim tree as ASCII art output.   

May be something like this:
```python
In [1]: model.components.pretty_print(entities=['name', 'bodies'],  levels=None, indent=4, sort_keys=False)
Out [1]: 
my_model
- comp1
    - comp1.1
        o my_surface_in_1.1
    o my_surface1
- comp2
    - comp2.1
        o my_surface_in_2.1-1
        o my_surface_in_2.1-2
    - comp2.2
        o my_surface_in_2.2
```
In this example "-" is used for components and "o" is used for bodies. The name is printed for components and bodies of all levels (default: levels=None).

RobPasMue, 2024-09-04:
Love this feature. I'll definitely work on it!

RobPasMue, 2024-09-04:
@dkowollik - could you have a look at https://github.com/ansys/pyansys-geometry/pull/1403#issuecomment-2329147856 and see if that would fit your needs? :)

dkowollik, 2024-09-04:
@RobPasMue looks fine for me. I would add this method to the component class as well, therefore you can start from different levels and go recursive down the tree from that location. If you decide to do so, I would probably add an extra line that prints the subtree location info of this component. Something like this related to your example: 
```python
In[1]: design.components[0].components[0].tree_print()
Out[1]:
location: Deletion_Test > Component_1 > Nested_1_Component_1
subtree:
(comp) Nested_1_Nested_1_Component_1
|---(body) nested_1_nested_1_comp_1_circle
```

RobPasMue, 2024-09-05:
Hi @dkowollik! Yes, it is implemented in that way. The "Component" class is the one that has the method implemented ("Design" is a class that extends "Component" - that is why I am able to call it from the ``design`` object.

I will implement the location concept, I think that's a very valid point! Thanks for the feedback :)

RobPasMue, 2024-09-05:
Some additional enhancements lead to this format @dkowollik - I think it's ready now for testing and merging =) Let me know your thoughts.

https://github.com/ansys/pyansys-geometry/pull/1403#issuecomment-2330864973

dkowollik, 2024-09-05:
I like the spacing. I think this is a good solution.

RobPasMue, 2024-09-05:
Awesome, thanks for the feedback! Moving forward with testing in that case :)

closed 2024-09-06 by RobPasMue