ansys/pyansys-geometry #1415: Add server exit to del operator and finished process

dkowollik, 2024-09-09:
### 📝 Description of the feature

Hi, 

can you please at the server exit to the python `del` operator and once a python process is finished? PyACP makes use of the Python stdlib `weakref.finalize`. The callback is executed, if the instance is not referenced anymore or if the python process ends.

dkowollik, 2024-09-09:
Hi @RobPasMue,

I understand. I am launching by the same python terminal. I meant something different. If I use a script and forget to implement the `modeler.exit()` or `modeler.close()`, the server keeps on running and blocks licenses. In PyMAPDL and PyACP they stop the server, even if you launched during this python process and forget to call exit or close.  
In PyACP `del modeler` would also close the server.

RobPasMue, 2024-09-09:
Aha, gotcha! For sure, let me implement that. =)

RobPasMue, 2024-09-09:
PR coming up @dkowollik - see #1417. This should make sure every time there is a client connection, it is closed.

PR #1417 (2024-09-10): feat: ensure GrpcClient class closure upon deletion
## Description
Ensuring the GrpcClient closure method gets called upon completion.

## Issue linked
Closes #1415

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [X] I have assigned this PR to myself.
- [X] I  [... cut]
closed 2024-09-10 by RobPasMue