ansys/pyansys-geometry #1452: How to install and run ANSYS Geometry Service/PygGeometry on Linux

c-eriksson86, 2024-10-04:
### Description of the modifications

Hey,
As I have understood it, pyGeometry uses ANSYS Geometry Service. ANSYS Geometry Service is available as a product flag when installing ansys silent. However, it is not supported on linux, i.e. the flag doesn't work there. 
So, how to install ANSYS Geometry Service and PyGeometry on Linux?
Am I missing something simple here?

RobPasMue, 2024-10-04:
Hi @c-eriksson86 - the Ansys Geometry Service is currently not available on Linux, despite the development team is working to make it available for Linux (i.e. it's part of the roadmap).

Closing issue!

c-eriksson86, 2024-10-04:
Hey,
Thanks for the quick reply!

RobPasMue, 2024-10-04:
Of course, anytime! I must say though, the library itself (PyAnsys Geometry) will have the same API for all the backends. Meaning that if you develop your script against a backend like the Windows version of the Ansys Geometry Service, once the Linux version is available it should work out of the box (no changes needed) =) Hope this helps!

Kind regards!

kmahajan-cadfem, 2025-09-03:
@RobPasMue, is this information correct for newest version as well?

according to the [documentation](https://geometry.docs.pyansys.com/version/stable/getting_started/local/index.html#launch-a-local-session), it is possible to launch the linux local geometry service.

we are looking for  2 things

1. unified installer for linux which packs geometry service with it, as per [documentation with flag](https://geometry.docs.pyansys.com/version/dev/getting_started/faq.html#how-is-the-ansys-geometry-service-installed)
2. launch the geometry service locally without the containers

kmahajan-cadfem, 2025-09-03:
sorry missed the last line while copy pasting

`GeometryExitedError: Geometry service connection terminated: Service.Start() has not been called.`

RobPasMue, 2025-09-03:
Seems like a licensing issue -- do you have the license environment variables defined?

kmahajan-cadfem, 2025-09-03:
which env variable should be set, I have tried following 
-  `ANSYSLIC_DIR`
-  `ANSYLMD_FILE`
- `LICENSE_SERVER`

[image]

[image]

RobPasMue, 2025-09-03:
> * `ANSYSLMD_LICENSE_FILE`
> * `LICENSE_SERVER`

These two should work yeah. Can we see the server logs on your side? You can use https://geometry.docs.pyansys.com/version/stable/api/ansys/geometry/core/modeler/Modeler.html#Modeler.get_service_logs AFAIK

kmahajan-cadfem, 2025-09-08:
@RobPasMue, anything else that I can try, to get it working?
is it still related to the licensing?

RobPasMue, 2025-09-09:
Maybe @b-matteo or @RyanJWard can help on this since they have troubleshooted licensing issues in the past

RyanJWard, 2025-09-09:
@kmahajan-cadfem What Linux distro are you using? You can try setting ANSYSCL252_DIR = /opt/ansys/v252/GeometryService/licensingclient

kmahajan-cadfem, 2025-09-09:
I am using the `Rocky Linux 8.10`

RyanJWard, 2025-09-09:
@kmahajan-cadfem @RobPasMue There is a current limitation where the service won't run on Rocky 8. Not sure if we should open a new issue to track this as a known limitation

kmahajan-cadfem, 2025-09-10:
Thank you very much for the information

could you please let me  know which distributions are tested and works without problems for the Geometry Service?

it would be great to have this information documented somewhere for future