ansys/pyansys-geometry #1453: How to use the modeler with geometry service

emjn2, 2024-10-04:
### Description of the modifications

When trying the documentation examples that use the launch_modeler() command I get an uncaught execution when creating a new design even though the connection is displayed as healthy - see below

I am using ANSYS 2024R1 and I have the geometry services installed. 

>>> modeler = launch_modeler()
>>> print(modeler)
Ansys Geometry Modeler (0x1e83da41e90)

Ansys Geometry Modeler Client (0x1e83c6ee4d0)
  Target:     localhost:49721
  Connection: Healthy
>>> design = modeler.create_design('test')
CRITICAL -  -  logger - handle_exception - Uncaught exception
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "Z:\.venv_test\Lib\site-packages\ansys\geometry\core\modeler.py", line 174, in create_design
    design = Design(name, self)
             ^^^^^^^^^^^^^^^^^^
  File "Z:\.venv_test\Lib\site-packages\ansys\geometry\core\errors.py", line 98, in wrapper
    raise GeometryExitedError(
ansys.geometry.core.errors.GeometryExitedError: Geometry service connection terminated: Api.Initialize has not been called.

RobPasMue, 2024-10-07:
Hi @emjn2 - thanks for opening this issue! I have a few questions beforehand...

1. Could you share the specific example you are trying to run?
2. Can you share as well the list of packages in your virtual environment? (i.e. output from ``pip list``)
3. Do you have other Ansys versions installed?

Let's try and take it from there.

emjn2, 2024-10-07:
Hi @RobPasMue 

Please see below, thanks

1. Any example in the documentation that requires launch_modeler e.g. 101: Modeling, NACA Aerofoil generator (both), etc. 
2. 
Package                             Version
----------------------------------- -----------
aiohappyeyeballs                    2.4.0
aiohttp                             3.10.5
aiosignal                           1.3.1
ansys-api-dbu                       0.3.4
ansys-api-geometry                  0.4.8
ansys-geometry-core                 0.7.2
ansys-tools-path                    0.6.0
ansys-tools-visualization-interface 0.4.4
appdirs                             1.4.4
attrs                               24.2.0
beartype                            0.18.5
certifi                             2024.8.30
charset-normalizer                  3.3.2
click                               8.1.7
colorama                            0.4.6
contourpy                           1.3.0
cycler                              0.12.1
flexcache                           0.3
flexparser                          0.3.1
fonttools                           4.53.1
frozenlist                          1.4.1
grpcio                              1.66.1
grpcio-health-checking              1.66.1
idna                                3.10
kiwisolver                          1.4.7
matplotlib                          3.9.2
more-itertools                      10.5.0
msgpack                             1.1.0
multidict                           6.1.0
numpy                               2.1.1
packaging                           24.1
pillow                              10.4.0
Pint                                0.24.3
pip                                 24.2
platformdirs                        4.3.6
pooch                               1.8.2
protobuf                            5.28.2
pyparsing                           3.1.4
python-dateutil                     2.9.0.post0
pyvista                             0.44.1
requests                            2.32.3
scipy                               1.14.1
scooby                              0.10.0
semver                              3.0.2
setuptools                          65.5.0
six                                 1.16.0
trame                               3.6.5
trame-client                        3.3.1
trame-server                        3.2.3
trame-vtk                           2.8.10
trame-vuetify                       2.7.1
typing_extensions                   4.12.2
urllib3                             2.2.3
vtk                                 9.3.1
websockets                          13.0.1
wheel                               0.44.0
wslink                              2.2.1
yarl                                1.11.1

3. I have v211, v231 and v241 but only v241 has the geometry services installed

RobPasMue, 2024-10-07:
I gave it a try on my side to the following:

* Using Ansys 24R1(in my case since I have more versions I had to adapt it as follows):

```py
modeler = launch_modeler(mode="geometry_service", product_version=241)
```

* Installing ``ansys-geometry-core==0.7.2`` on a clean virtual environment
* Running example: https://geometry.docs.pyansys.com/version/stable/examples/04_applied/01_naca_airfoils.html

Everything worked as expected for me which makes me think that there must be some kind of issue on licensing... Could you navigate to the following folder? (``C:\Users\Public\Documents\Ansys\GeometryService``) This folder contains the log files from the service (starting on version 0.7.2 in fact, which is your current version).

Could you also share the content of the latest log file you have? Please scrape any private IP/domain you might not be willing to share from the log - licensing gets logged in it as well.

emjn2, 2024-10-07:
Contents from my latest log file is shown below - looks as through there is an issue with setting the licence server
 ```bash
Licensing: ANSYS241_DIR already set to 'C:\Program Files\ANSYS Inc\v241\ANSYS'
Licensing: Server.SetLicenseServer() has not been called.
Licensing: License checkout failed
gRPC services subscription done.
Server started. Listening on endpoint: localhost:XXXX
```

RobPasMue, 2024-10-07:
Aha... makes sense. Seems like your license server is not set. Please make sure you have a license server configured which you can connect to

[image]

Make sure you also have a Discovery Modeling License - just so you know, the Mechanical Enterprise license won't be enough 😢

emjn2, 2024-10-07:
Checking the 2024 R1 Ansys Capabilities Chart I see that Discovery Modeling uses the same licence as SpaceClaim 

[image]

Our licence server is on a remote host and using the below works. This launches SpaceClaim which my understanding is not as efficient as using the geometry_service directly. 

```
modeler = launch_modeler('spaceclaim')
```

RobPasMue, 2024-10-07:
> Our licence server is on a remote host and using the below works. This launches SpaceClaim which my understanding is not as efficient as using the geometry_service directly.
> ```
> modeler = launch_modeler('spaceclaim')
> ```

Agreed - the Ansys Geometry Service is more efficient than launching SpaceClaim. Have you tried launching SpaceClaim though? I would be surprised you don't hit the same licensing issue nonetheless.

To me, it looks like the license server is not properly configured in your machine. Could you define an environment variable in your system that points to the license server and is called as follows?

``ANSYSLMD_LICENSE_FILE=1055@"$LICENSE_SERVER"``

1055 is the default port for license servers - if you have a different port, please adapt.

emjn2, 2024-10-07:
No luck I'm afraid. The log file still states that the Server.SetLicenseServer has not been called. When I use spaceclaim it does not generate a log file. I have no issue using PyTurboGrid or PyFluent with how I have the license server setup

RobPasMue, 2024-10-08:
This is very strange... could you try running all this in an "admin" terminal just in case?

Also, a second option is... could you try executing directly the following EXE? (C:\Program Files\ANSYS Inc\v241\GeometryService\Presentation.ApiServerDMS.exe``)

The log should show something like this...

[image]

emjn2, 2024-10-08:
I tried executing the exe directly and got a strange output

[image]

I'm also in contact with our supplier to confirm that we have a suitable licence for the product

Regards

RobPasMue, 2024-10-08:
Can you directly double-click on the executable from the file explorer rather than running it through CLI? Sorry for all the forth and back. Let's see if the supplier can confirm you have the required license.

emjn2, 2024-10-08:
Double-clicking on the exe temporarily opens a window. It is too quick to read what it contains but I would guess it is the unhandled exception.

emjn2, 2024-10-08:
Our ansys supplier has confirmed we do have a licence for Discovery Modeling. I tried launching Discovery directly and I get the following error - could this be the cause behind my issues here?

[image]

RobPasMue, 2024-10-08:
The way I see it is that there seems to be an issue with the license for sure. If you can't launch Discovery either, it might be the same problem behind the scenes

emjn2, 2024-10-17:
Hi,

I've been looking into this with my colleague who is also having similar issues with geometryservice. 

He has Discovery installed and is working on his machine, unlike mine - I think it is a graphics card issue.

Before installing geometryservice he ran the following command which launched a spaceclaim session in the background

```
modeler = launch_modeler()
design = modeler.create_design('test')
```

After installing geometryservice using the procedure detailed here (https://github.com/ansys/pyansys-geometry/discussions/1374) the above command now fails with the critical error that I originally posted.

It looks as though pyGeometry knows that geometry service is installed but it is not working properly. All other ANSYS products can see and access the licence server.

RobPasMue, 2024-10-17:
Hi @emjn2 - thanks for the update. One question... are you using Ansys 24R2 or Ansys 24R1? Your initial message is referring to Ansys 24R1 but procedure on #1374 is specific for 24R2

RobPasMue, 2024-10-17:
Just FYI - ``launch_modeler()`` has an automated scheme for selecting the backend. If you don't pass in a "mode" it will attempt to:

1. Launch the Geometry Service
2. If 1 failed, launch SpaceClaim
3. If 2 failed, launch Ansys Discovery

RobPasMue, 2024-10-17:
Okay thanks for clarifying! Let's do something... can you execute your PyAnsys Geometry script from and "admin privileged" terminal and see if it still fails?

I'm not sure if this: https://github.com/ansys/pyansys-geometry/issues/1453#issuecomment-2399070924 was executed from and admin terminal or not.

I will investigate on my side as well on a clean VM... but like I reported here: https://github.com/ansys/pyansys-geometry/issues/1453#issuecomment-2396641032 I didn't face any issues on my side...

RobPasMue, 2024-10-17:
Please update to the latest PyAnsys Geometry version as well (0.7.4 currently)

emjn2, 2024-10-17:
I've just created a clean virtual environment with the latest version of PyAnsys Geometry within an administrator terminal.

Unfortunately I get the same result when trying geometry service (see below) and it works with SpaceClaim

[image]

jmabshof, 2025-03-07:
@RobPasMue is this a user-config issue, or a real bug?

RobPasMue, 2025-04-09:
Hi @jmabshof -- I think it's a user configuration problem honestly. But it doesn't have to be the user's "fault" -- it might be related to restrictions on his network or similar.. We haven't experienced issues with other users though. Also, the user was trying to usse Ansys 24R1. Maybe these issues have been resolved with the new releases. I'd propose it for closing if you don't mind.