ansys/pyansys-geometry #1533: run_discovery_script_file accepts only non list data, flat dict

mcMunich, 2024-11-12:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

if I try to use the run_discovery_script_file to send anything other than individual string parameters, it fails. While this is slightly documented https://geometry.docs.pyansys.com/version/dev/api/ansys/geometry/core/modeler/Modeler.html#Modeler.run_discovery_script_file() it doesn't say that the dict can be exclusively flat dict.  I have tried all variations (nested dictionaries, tuples, lists  of strings etc) and only a key:string combination works for the script_args sent to a discovery file. 
And, to make it useful in the future, it would be nice to have some more inputs available. 

### 📝 Steps to reproduce

# run the following script to try to send a dict with two keys,  the second key being a list of strings

```
import tempfile
from ansys.geometry.core import launch_modeler_with_discovery
import os

version = 242

file_name = 'dummy.py'
tempfile.gettempdir()
path2dummy = os.path.join(tempfile.gettempdir(),file_name)

with open(path2dummy,'w') as f:
    f.write('#nothing')

input_dict = {'a':1, 'b':['1','2']}

modeler_discovery = launch_modeler_with_discovery(product_version = version)

modeler = modeler_discovery

values = modeler.run_discovery_script_file(path2dummy, script_args = input_dict)

modeler.close()
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

242 and/or 251

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
ni
```

RobPasMue, 2024-11-12:
I completely agree - let me see if we can do something here to have it working.

Alberto-DM, 2024-12-19:
The limitation is known. 
I'm using this workaround to overcome the issue.

First I serialize every value in the args dict. 
If args is the dictionary that I want to pass to the script, I will actually pass new_args:

```
new_args = copy.deepcopy(args)
for k, v in new_args.items():
    new_args[k] = json.dumps(v)
```

Then, **inside the script** that I will run, I add this lines at the beginning of the file to deserialize the arguments:

```
import json
keys = list(argsDict.Keys)
for k in keys:
    argsDict[k] = json.loads(argsDict[k])
```

In this way, the args values can be any structure that can be sterilized by `json.dumps`

I am actually doing this programmatically in a wrapper function of `modeler.run_discovery_script_file` so that I can directly pass any arg without worrying of its convertion.
You can also do the same for the result dictionary returned by the method (serialize inside the script and de-serialize outside).

@RobPasMue maybe a similar approach can be implemented natively in the method itself.

mcMunich, 2025-02-04:
thanks @Alberto-DM it works and is pretty smart. Maybe they can wrap that into pygeo in the near future or come up with a solution.

RobPasMue, 2025-02-05:
I like the approach! Let me talk with the team and see how we can add the service side without having users to add it in their SpaceClaim script

simondhawk, 2025-02-20:
Hello, I've tried the suggested workaround but unfortunately it is not working for me.
I'm running discovery 242, python 3.12.

Here below the steps to reproduce it.

import copy
import json
from ansys.geometry.core import launch_modeler_with_discovery

my_dict = {'i_parameter1' : '1' , 'i_parameter2' : '2'}  #(45 parameters tot)

new_args = copy.deepcopy(my_dict)
for k, v in new_args.items():
    new_args[k] = json.dumps(v)

modeler=launch_modeler_with_discovery(product_version=242)

design = modeler.open_file(geom_file)

result = modeler.run_discovery_script_file(my_script.py, script_args =new_args)

modeler.close()

my_script.py is-->

import json
keys = list(argsDict.keys())
for k in keys:
    argsDict[k] = json.loads(argsDict[k])

Alberto-DM, 2025-02-21:
> Hello, I've tried the suggested workaround but unfortunately it is not working for me. I'm running discovery 242, python 3.12.

Can you share the error message? 
Also, can you share the complete script?
Thank you

simondhawk, 2025-02-21:
Hi Alberto,
in attachment my scripts in order to reproduce the issue.
I've done one attempt using spaceclaim and another one using discovery.
Many thanks in advance.

[error_sc.txt](https://github.com/user-attachments/files/18907633/error_sc.txt)

[error_ds.txt](https://github.com/user-attachments/files/18907635/error_ds.txt)

[test_launch_discovery_script.zip](https://github.com/user-attachments/files/18907636/test_launch_discovery_script.zip)

Alberto-DM, 2025-02-21:
I will look in to it. 
In the meanwhile, I've spotted the following things. 
1. In both cases (SC and Disco) you are not passing the args dict to the `run_discovery_script_file` method. In your case, you have to add the argument `script_args=new_args`.
2. Discovery script fails on line 19 (`modeler.open_file(file_path=r'{}'.format(geom_path))`). The `run_discovery_script_file` is commented.