ansys/pyansys-geometry #1579: Enable repair tools in core services

RobPasMue, 2024-12-10:
Core services seems to have catched up with repair tools. Let's activate them and verify that the tests are passing.

jonahrb, 2025-02-03:
@RobPasMue is this considered complete now?

RobPasMue, 2025-02-04:
Some tests are still being skipped and require to be fixed before we can consider closing it. Search for "skip_if_core_service" in the tests folder. I am hitting the following modules:

test_geometry_commands
test_prepare_tools
test_repair_tools
test_measurement_tools

I believe all the skips in those modules should be solved before we can officially say that we support the tools (not only repair but also prepare and measurement)

jonahrb, 2025-02-13:
@RobPasMue @jacobrkerstetter i believe Jacob's PR will close this issue - bringing measurement tools?

jonahrb, 2025-02-13:
The only thing remaining after that is find/fix small faces - the behavior is different between DMS and CoreService, but that can still close this issue and open a separate one

RobPasMue, 2025-02-13:
Yes that's the only remaining one. Is the behavior different in the sense that we get different results? Or is the API not implemented?

RobPasMue, 2025-02-13:
We might have to adapt the tests in case it's just different results so that they accommodate both backends. I can understand that the results might not be exactly the same.

PR #1580 (2025-01-10): feat: enable (partially) prepare and repair tools in Core Service
## Description
Enabling prepare and repair tools in Core Service

## Issue linked
Closes #1579 

## Checklist
- [ ] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [X] I have assigned this PR to myself.
- [X] I hav [... cut]
closed 2025-02-14 by RobPasMue