ansys/pyansys-geometry #1622: body.imprint_projected_curves() not working as expected

sp-claim, 2025-01-13:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Hi,

I am using Ansys geometry service 2024R2 with Python 3.13 and ansys-geometry-core 0.76 & PrimeMesh. Everything has been installed on a clean environment. 
I have the following problem:

I created a body and tried projecting a sketch onto it. The sketch consits of a single circle. But if I try to project onto the body it results in just a line. 

If I create a surface from that sketch everything looks normal. It is unclear if its a bug right now. 

### 📝 Steps to reproduce

from ansys.geometry.core import launch_modeler_with_spaceclaim
from ansys.geometry.core import launch_modeler_with_geometry_service
from ansys.geometry.core import launch_modeler
from ansys.geometry.core.materials import Material, MaterialProperty, MaterialPropertyType
from ansys.geometry.core.math import UNITVECTOR3D_Z, Frame, Plane, Point2D, Point3D, UnitVector3D, UNITVECTOR3D_X, \
    UNITVECTOR3D_Y
from ansys.geometry.core.misc import UNITS
from ansys.geometry.core.sketch import Sketch
from ansys.geometry.core.misc import DEFAULT_UNITS, Distance

modeler = launch_modeler(mode="spaceclaim",version=242)

pxy=Plane(origin=Point3D([0,0,-100],unit=UNITS.mm),direction_x=UNITVECTOR3D_X,direction_y=UNITVECTOR3D_Y)
s1=Sketch(pxy)
s1.box(Point2D([1550/2,0],unit=UNITS.mm),1550,100)
design=modeler.create_design("smirror")
mbody=design.extrude_sketch("mb",sketch=s1,distance=Distance(200,unit=UNITS.mm))
cylplane=Plane(origin=Point3D([0,0,0],unit=UNITS.mm),direction_x=UNITVECTOR3D_X,direction_y=UNITVECTOR3D_Z)
s2=Sketch(cylplane)
s2.circle(center=Point2D([50,0],unit=UNITS.mm),radius=Distance(10,unit=UNITS.mm))
face=design.create_surface("cyl",sketch=s2)
testface=mbody.imprint_projected_curves(direction=UNITVECTOR3D_Y,sketch=s2,closest_face=True)

design.export_to_scdocx()
modeler.close()

[image]

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024R2

### 🐍 Which Python version are you using?

3.13

### 📦 Installed packages

```shell
aiohappyeyeballs==2.4.4
aiohttp==3.11.11
aiosignal==1.3.2
ansys-api-dbu==0.3.6
ansys-api-geometry==0.4.16
ansys-geometry-core==0.7.6
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.7.0
attrs==24.3.0
beartype==0.19.0
certifi==2024.12.14
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
flexcache==0.3
flexparser==0.4
fonttools==4.55.3
frozenlist==1.5.0
grpcio==1.67.1
grpcio-health-checking==1.67.1
idna==3.10
kiwisolver==1.4.8
matplotlib==3.10.0
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
numpy==2.2.1
packaging==24.2
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pooch==1.8.2
propcache==0.2.1
protobuf==5.29.3
pyparsing==3.2.1
python-dateutil==2.9.0.post0
pyvista==0.44.1
requests==2.32.3
scipy==1.15.0
scooby==0.10.0
semver==3.0.2
six==1.17.0
trame==3.7.6
trame-client==3.5.1
trame-server==3.2.7
trame-vtk==2.8.14
trame-vuetify==2.8.0
typing_extensions==4.12.2
urllib3==2.3.0
vtk==9.4.1
websockets==14.1
wslink==2.2.2
yarl==1.18.3
```

sp-claim, 2025-01-14:
@RobPasMue Projecting the circle in Z-Direction works. The orientation of the plane the sketch is without any impact. 

XZ Plane -->Y-Projection--> Line on faces with Y Normal vector
XY Plane -->Y-Projection--> Line on faces with Y Normal vector
ZY Plane -->Y-Projection--> Line on faces with Y Normal vector

XY Plane -->Z-Projection--> circle on face with Z-Normal vector
XZ Plane -->Z-Projection-->circle on face with Z-Normal vector
YZ Plane -->Z-Projection-->circle on face with Z-Normal vector

This seems odd to me. Seems like the sketch orientation is always the same after passing the object to the function.

jonahrb, 2025-02-03:
@RobPasMue @sp-claim i believe found the issue here, the plane is hardcoded on the backend to be the XY plane, this was an oversight for an original workflow. We must change the ImprintCurves, ProjectCurves, and ImprintProjectedCurves proto's to include a plane and use the provided plane.

RobPasMue, 2025-02-04:
FYI @sp-claim - we are working on a fix for this but sadly it won't be available on your side earlier than 25R2 since it implies modifying the server shipped with the Ansys installation. Sorry for the inconveniences!

sp-claim, 2025-02-04:
> [Verfolgungsbild][https://github.com/notifications/beacon/BOOH3RQR4O7V533UY6HGXLL2OCSBNA5CNFSM6AAAAABVCXEOJWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU47KJHE.gif]

Thx for the info. The same happened when using spaceclaim it's not only the server.

RobPasMue, 2025-02-05:
Hi @sp-claim! The server we mention is shared by all products (SpaceClaim, Discovery and the Geometry Service). Consequently, it will also be fixed on SpaceClaim. Thanks for bringing the problem to our attention!