ansys/pyansys-geometry #1724: Boolean Intersection not working in all cases

rmanno91, 2025-02-05:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

The boolean intersection is not working correctly. 
A script to reproduce the bug is reported herein.

red_body.intersect(green_body) -> Exception: Operation of type 'intersect' failed: bodies do not intersect.
green_body.intersect(red_body) -> works fine
[image]

### 📝 Steps to reproduce

```python
from ansys.geometry.core import launch_modeler_with_geometry_service
from ansys.geometry.core.math import Point2D, Point3D, Plane, UNITVECTOR3D_X, UNITVECTOR3D_Y
from ansys.geometry.core.misc import DEFAULT_UNITS
from ansys.geometry.core.sketch import Sketch

wx = 10
wy = 10
wz = 2
radius = 1
unit = DEFAULT_UNITS.LENGTH

modeler = launch_modeler_with_geometry_service()
design = modeler.create_design("Test")

start_at = Point3D([wx / 2, wy / 2, 0.0], unit=unit)

plane = Plane(
    start_at,
    UNITVECTOR3D_X,
    UNITVECTOR3D_Y,
)
box_plane = Sketch(plane)
box_plane.box(Point2D([0.0, 0.0], unit=unit), width=wx, height=wy)

box = design.extrude_sketch("box", box_plane, wz)

point = Point3D([wx / 2, wx / 2, 0.0], unit=unit)
plane = Plane(point, UNITVECTOR3D_X, UNITVECTOR3D_Y)
sketch_cylinder = Sketch(plane)
sketch_cylinder.circle(Point2D([0.0, 0.0], unit=unit), radius=radius)
cylinder = design.extrude_sketch("cylinder", sketch_cylinder, wz)
cylinder.intersect(box)
#it works if you do 
#box.intersect(cylinder)
# design.plot(show_edges=True)
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

25r2

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
accessible-pygments==0.0.5
aiohappyeyeballs==2.4.4
aiohttp==3.11.11
aiosignal==1.3.2
alabaster==1.0.0
annotated-types==0.7.0
ansys-api-dbu==0.3.12
ansys-api-geometry==0.4.35
ansys-api-mapdl==0.5.2
ansys-api-meshing-prime==0.1.4
ansys-api-platform-instancemanagement==1.1.2
ansys-geometry-core @ git+https://github.com/ansys/pyansys-geometry@9368e142bf60523467dc49801c9e2214d69a1300
ansys-mapdl-core @ git+https://github.com/ansys/pymapdl@d2a47874b1bed613a928494c320a1db10db6544e
ansys-mapdl-reader==0.54.2
ansys-materials-manager @ git+https://github.com/ansys/pymaterials-manager@044c739c6e8f62bbb8dcea203f9cef44b371dd5d
ansys-math-core==0.2.2
ansys-meshing-prime @ git+https://github.com/ansys/pyprimemesh@4a3390eb14287f5cd8e080ad6d857797458dfe0d
ansys-platform-instancemanagement==1.1.2
ansys-sphinx-theme==1.2.6
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.1
appdirs==1.4.4
attrs==24.2.0
babel==2.17.0
beartype==0.19.0
beautifulsoup4==4.13.3
build==1.2.2.post1
certifi==2025.1.31
cfgv==3.4.0
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
coverage==7.6.10
cycler==0.12.1
distlib==0.3.9
docutils==0.21.2
filelock==3.17.0
flexcache==0.3
flexparser==0.4
flit==3.10.1
flit_core==3.10.1
fonttools==4.55.8
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.62.3
id==1.5.0
identify==2.6.6
idna==3.10
imagesize==1.4.1
importlib_metadata==8.6.1
iniconfig==2.0.0
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.1.0
Jinja2==3.1.5
keyring==25.6.0
kiwisolver==1.4.8
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
mdurl==0.1.2
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
nh3==0.2.20
nodeenv==1.9.1
numpy==2.2.2
numpydoc==1.8.0
packaging==24.2
pdf2image==1.17.0
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pluggy==1.5.0
pooch==1.8.2
pre_commit==4.1.0
propcache==0.2.1
protobuf==4.25.6
psutil==6.1.1
pyansys-tools-versioning==0.6.0
pydantic==2.10.6
pydantic_core==2.27.2
pydata-sphinx-theme==0.16.1
Pygments==2.19.1
pyiges==0.3.1
pyparsing==3.2.1
pyproject_hooks==1.2.0
pytest==8.3.4
pytest-cov==6.0.0
python-dateutil==2.9.0.post0
pyvista==0.44.2
pywin32-ctypes==0.2.3
PyYAML==6.0.2
readme_renderer==44.0
requests==2.32.3
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.9.4
scipy==1.15.1
scooby==0.10.0
semver==3.0.4
six==1.17.0
snowballstemmer==2.2.0
soupsieve==2.6
Sphinx==8.1.3
sphinx-copybutton==0.5.2
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
tabulate==0.9.0
tomli_w==1.2.0
tqdm==4.67.1
trame==3.8.0
trame-client==3.5.2
trame-server==3.3.0
trame-vtk==2.8.14
trame-vuetify==2.8.1
twine==6.1.0
typing_extensions==4.12.2
urllib3==2.3.0
virtualenv==20.29.1
vtk==9.3.1
websockets==14.2
wheel==0.45.1
wslink==2.2.2
yarl==1.18.3
zipp==3.21.0
```

smereu, 2025-02-05:
The implementation of intersect has been changed from using the Space Claim core APIs to commands.  The reason is that intersect (as well as merge/unite and subtract) using Core APIs only work for bodies in the same part.  The current implementation was trying to overcome this limitation by making copies but it has issues with leaving bodies behind and with transformations.  For this reason we decided to switch to using commands. If you use Discovery/SC, the implementation we exposed is what you get from the Combine tool from the Design Tab.  This behaviorof Intersect is a little different in the sense for your case it gives you the cylinder (which is actually renamed solid, because it is no longer the existing cylinder) and also the hollow box.  The boolean operation from CoreAPIs was only leaving the cylinder.  As you noted, there is another difference. In the Core APIs, box intersect cylinder and cylinder intersect box both give you the same result. With the command, the second doesn't work. It does work however, if you make the cylinder stick up a little bit. In other words, the case where you try to intersect with a body completely inside another one requires the intersect to be done from the "containing" body.  If you feel that there has been a loss of functionality that is preventing you from doing what you want, please contact me.  Before we make any change though, we have to gather all the requirements and expectations. I hope that the explanation helped!

RobPasMue, 2025-02-06:
IMO - the previous boolean operations and the new way they are implemented might require a revisit... I think we should keep them completely independent and leave the command-based implementation inside the geometry commands module. This will allow users to benefit from the implementation that they prefer.

PR #1725 (2025-02-11): fix: revert boolean ops logic and hold-off on commands-based implementation (temporarily)
## Description
Coming from #1724 

## Issue linked
Closes #1724

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [X] I have assigned this PR to myself.
- [X] I have added  [... cut]
closed 2025-02-11 by RobPasMue