ansys/pyansys-geometry #1735: Simba App build fails on ARM because of VTK dependency

ans-spohekar, 2025-02-10:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Unable to find installation candidate for vtk (9.3.1) on ARM machines

[image]

cc: @V-Dandekar @h-krishnan

### 📝 Steps to reproduce

Try to install ansys-geometry-core on ARM machine

### 💻 Which operating system are you using?

Linux

### 📀 Which ANSYS version are you using?

252

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
accessible-pygments==0.0.5
aiohappyeyeballs==2.4.6
aiohttp==3.11.12
aiosignal==1.3.2
alabaster==1.0.0
altgraph==0.17.4
annotated-types==0.7.0
ansys-api-dbu==0.3.11
ansys-api-fluent==0.3.35
ansys-api-geometry==0.4.33
ansys-api-meshing-prime==0.1.5.dev1
ansys-api-platform-instancemanagement==1.1.2
ansys-api-tools-filetransfer==0.1.1
ansys-fluent-core==0.29.0
ansys-geometry-core==0.8.2
ansys-meshing-prime==0.7.0a1.dev28
ansys-platform-instancemanagement==1.1.2
-e git+https://github.com/ansys-internal/simba-app.git@170f40718293f19eb0c27b22432a74417bcf9be7#egg=ansys_simba_app
ansys-simba-core==0.44.1
ansys-simba-plugin-fluent==0.10.2
ansys-simba-plugin-geometry==0.3.0
ansys-simba-plugin-prime==0.19.0
ansys-simba-settings==0.19.0
ansys-sphinx-theme==1.2.4
ansys-tools-filetransfer==0.1.1
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.1
ansys-units==0.3.4
anyio==4.8.0
appdirs==1.4.4
async-timeout==5.0.1
attrs==24.2.0
autoflake==2.3.1
babel==2.16.0
backports.tarfile==1.2.0
beartype==0.19.0
beautifulsoup4==4.12.3
build==1.2.2.post1
CacheControl==0.14.1
cachetools==5.5.0
certifi==2025.1.31
chardet==5.2.0
charset-normalizer==3.4.1
cleo==2.1.0
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
coverage==7.6.10
crashtest==0.4.1
cycler==0.12.1
deepmerge==2.0
distlib==0.3.9
dnspython==2.7.0
docker==7.1.0
docutils==0.21.2
dulwich==0.21.7
exceptiongroup==1.2.2
fastapi==0.115.8
fastjsonschema==2.21.1
filelock==3.16.1
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
frozenlist==1.5.0
geomdl==5.3.1
googleapis-common-protos==1.66.0
grpcio==1.67.1
grpcio-health-checking==1.62.3
grpcio-status==1.62.3
h11==0.14.0
httpcore==1.0.7
httpx==0.28.1
idna==3.10
imagesize==1.4.1
importlib_metadata==8.6.1
iniconfig==2.0.0
installer==0.7.0
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.1.0
Jinja2==3.1.5
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
keyring==25.6.0
kiwisolver==1.4.8
lxml==5.3.1
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
mdurl==0.1.2
mistune==3.1.0
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
nh3==0.2.20
nltk==3.9.1
numpy==2.2.2
numpydoc==1.8.0
packaging==24.2
pandas==2.2.3
pdf2image==1.17.0
pefile==2023.2.7
pexpect==4.9.0
picobox==4.0.0
pillow==11.1.0
Pint==0.24.4
pkginfo==1.10.0
platformdirs==4.3.6
pluggy==1.5.0
poetry==1.8.5
poetry-core==1.9.1
poetry-plugin-export==1.8.0
poetry-pyinstaller-plugin==1.3.0
pooch==1.8.2
propcache==0.2.1
protobuf==4.25.6
ptyprocess==0.7.0
pyansys-tools-report==0.8.1
pydantic==2.10.6
pydantic_core==2.27.2
pydata-sphinx-theme==0.15.4
pyflakes==3.2.0
Pygments==2.18.0
pyinstaller==6.12.0
pyinstaller-hooks-contrib==2025.1
pyparsing==3.2.1
pyproject-api==1.8.0
pyproject_hooks==1.2.0
pytest==8.3.4
pytest-aiohttp==1.0.5
pytest-asyncio==0.25.1
pytest-cov==6.0.0
pytest-timeout==2.3.1
python-dateutil==2.9.0.post0
pytz==2024.2
pyvista==0.44.2
pywin32==308
pywin32-ctypes==0.2.3
PyYAML==6.0.2
RapidFuzz==3.11.0
readme_renderer==44.0
referencing==0.35.1
regex==2024.11.6
requests==2.32.3
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.9.4
rpds-py==0.22.3
scipy==1.15.1
scooby==0.10.0
semver==3.0.4
shellingham==1.5.4
six==1.17.0
sniffio==1.3.1
snowballstemmer==2.2.0
soupsieve==2.6
Sphinx==8.1.3
sphinx-copybutton==0.5.2
sphinx_design==0.6.1
sphinx_mdinclude==0.6.2
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-httpdomain==1.8.1
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-openapi==0.8.4
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
starlette==0.45.3
tabulate==0.9.0
tomli==2.2.1
tomlkit==0.13.2
tox==4.23.2
tqdm==4.67.1
trame==3.8.0
trame-client==3.5.2
trame-server==3.3.0
trame-vtk==2.8.15
trame-vuetify==2.8.1
trove-classifiers==2024.10.21.16
twine==5.1.1
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.3.0
uvicorn==0.34.0
virtualenv==20.28.0
vtk==9.3.1
websockets==14.2
wslink==2.2.2
yarl==1.18.3
zipp==3.21.0
```

RobPasMue, 2025-02-10:
@ans-spohekar -- this is not a bug, but rather an enhancement request. You are requesting PyAnsys Geometry to be compatible with ARM builds. Currently, due to its dependencies, it is not. The only thing we can do is completely decouple the graphics dependencies from the default dependencies of the package. But again -- this is not a bug, it's a feature request.

dnwillia-work, 2025-02-19:
@RobPasMue, we've already been engaging with Kitware to provide VTK support for ARM and have already tried out building it on the GitHub runners that seems to be working.

MaxJPRey, 2025-02-19:
@dnwillia-work Do you know when that could be available/used?

closed 2025-02-27 by RobPasMue

dnwillia-work, 2025-03-05:
@MaxJPRey I'm not certain of the exact date but their team is looking at it, so I'd say sometime this year.  @ahernsean might be know or be able to find out more concrete timing.

RobPasMue, 2025-03-05:
Maybe not released yet but it's been merged into their main branch I believe: https://github.com/pyvista/pyvista/issues/6731#issuecomment-2687911824

PR #1782 (2025-02-27): feat: separate graphics target
## Description
Making graphics an optional target inside PyAnsys Geometry. If users now want graphics features they can install the package as follows:

``pip install ansys-geometry-core[graphics]``

## Issue linked
Closes #1735 #1761

## Checklist
- [X] I have tested my changes locally.
- [x] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked th [... cut]