ansys/pyansys-geometry #1761: Visualization capability needs to be an optional feature

dnwillia-work, 2025-02-19:
### 📝 Description of the feature

Can the following forced dependencies be removed and made optional?

ansys-tools-visualization-interface, pyvista, vtk

The thing is that if we work on an application that has it's own native visuzalization capability then we don't need any of these and it clutters up our dependency tree.

### 💡 Steps for implementing the feature

Not 100% sure, but normally you move them to optional and make sure the dependent capabilities are checking if they are available or not.

RobPasMue, 2025-02-19:
Hi @dnwillia-work! Working on it, this is _almost_ a duplicate of #1735 - working on it actively. I'll try to get it done by next week. I completely agree on having an optional graphics target. The main problem (right now) is that the user of graphics is very coupled with some features... But we will get that solved!

dnwillia-work, 2025-02-24:
OK awesome thank you @RobPasMue.  This is why we split the pyfluent-visualization repo out quite some time ago from PyFluent itself.

closed 2025-02-27 by RobPasMue