ansys/pyansys-geometry #1765: Error on opening a cad file with size ~280 MB

ans-spohekar, 2025-02-21:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

If we open a file having size more than 275 MB it fails with bellow error

[image]

### 📝 Steps to reproduce

Try to import file with size more than 275 MB

```
from ansys.geometry.core import launch_modeler_with_geometry_service
modeler = launch_modeler_with_geometry_service()
design = modeler.open_file(path)
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

252

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
$ pip freeze
aiohappyeyeballs==2.4.4
aiohttp==3.11.11
aiosignal==1.3.2
ansys-api-dbu==0.3.12
ansys-api-geometry==0.4.38
ansys-geometry-core==0.9.0
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.1
attrs==25.1.0
beartype==0.19.0
certifi==2024.12.14
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
dataclasses-json==0.6.7
Deprecated==1.2.18
flexcache==0.3
flexparser==0.4
fonttools==4.55.7
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.67.1
idna==3.10
iniconfig==2.0.0
kiwisolver==1.4.8
marshmallow==3.26.1
matplotlib==3.10.0
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
mypy-extensions==1.0.0
numpy==2.2.2
packaging==24.2
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pluggy==1.5.0
pooch==1.8.2
propcache==0.2.1
protobuf==5.29.3
pygltflib==1.16.3
pyparsing==3.2.1
pytest==8.3.4
python-dateutil==2.9.0.post0
pyvista==0.44.2
requests==2.32.3
scipy==1.15.1
scooby==0.10.0
semver==3.0.4
six==1.17.0
trame==3.7.6
trame-client==3.5.2
trame-server==3.3.0
trame-vtk==2.8.14
trame-vuetify==2.8.1
typing-inspect==0.9.0
typing_extensions==4.12.2
urllib3==2.3.0
vtk==9.3.1
websockets==14.2
wrapt==1.17.2
wslink==2.2.2
yarl==1.18.3
```

RobPasMue, 2025-02-24:
We can set a "dynamic" channel size as a workaround. But the actual solution would be to allow the chunking and streaming of files in smaller pieces. I will work on the workaround and we will open an issue to implement the actual, robust solution.

RobPasMue, 2025-02-25:
The actual fix will be to allow the upload/download of files using streaming

RobPasMue, 2025-02-25:
Closing, since #1778 was created for the actual enhancement

PR #1779 (2025-03-16): feat: Streaming upload support
## Description
Client side support for streaming files to/from the server. 
Client side support for streaming tessellation from the server. 

## Issue linked
Closes  #1765, #1778 
## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate unit tests.
- [ ] I have reviewed my changes before submitting this pull request.
- [ ] I have linked the issue or issues that are solved to the PR if any.
- [  [... cut]
closed 2025-02-25 by RobPasMue