ansys/pyansys-geometry #1774: share_topology fails to write info when exported to scdocx/stride

ans-spohekar, 2025-02-25:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

If we apply share topology operation and export file to scdocx/stride, The 'shared topology' operation is not saved

### 📝 Steps to reproduce

1. Run following script
```python
from ansys.geometry.core import launch_modeler_with_core_service
from ansys.geometry.core.misc.options import ImportOptions
from ansys.geometry.core.misc import auxiliary

modeler = launch_modeler_with_core_service()
importOptions = ImportOptions(import_names=True)
design = modeler.open_file(file_path=path,import_options=importOptions)
bodies = auxiliary.get_all_bodies_from_design(design)
print(f"Apply share topology {modeler.prepare_tools.share_topology(bodies)}")
```

2. Import the file in Spaceclaim still it shows, design needs to apply share topology

[image]

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

252

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
accessible-pygments==0.0.5
aiohappyeyeballs==2.4.4
aiohttp==3.11.12
aiosignal==1.3.2
alabaster==1.0.0
altgraph==0.17.4
annotated-types==0.7.0
ansys-api-dbu==0.3.12
ansys-api-geometry==0.4.38
ansys-geometry-core==0.9.0
ansys-simba-core==0.46.1
-e git+https://github.com/ansys-internal/simba-plugin-geometry.git@90d91c4e00ce2362472d3233019c96bcdea8e2c5#egg=ansys_simba_plugin_geometry
ansys-simba-settings==0.20.0
ansys-sphinx-theme==1.2.6
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.3
anyio==4.8.0
attrs==24.2.0
autoflake==2.3.1
babel==2.16.0
beartype==0.19.0
beautifulsoup4==4.12.3
build==1.2.2.post1
CacheControl==0.14.2
cachetools==5.5.1
certifi==2024.12.14
cfgv==3.4.0
chardet==5.2.0
charset-normalizer==3.4.1
cleo==2.1.0
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
coverage==7.6.10
crashtest==0.4.1
cycler==0.12.1
dataclasses-json==0.6.7
Deprecated==1.2.18
distlib==0.3.9
dnspython==2.7.0
docker==7.1.0
docutils==0.21.2
dulwich==0.22.7
execnet==2.1.1
fastapi==0.115.8
fastjsonschema==2.21.1
filelock==3.17.0
flexcache==0.3
flexparser==0.4
fonttools==4.55.6
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.67.1
h11==0.14.0
httpcore==1.0.7
httpx==0.28.1
id==1.5.0
identify==2.6.6
idna==3.10
imagesize==1.4.1
importlib_metadata==8.6.1
iniconfig==2.0.0
installer==0.7.0
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.1.0
Jinja2==3.1.5
keyring==25.6.0
kiwisolver==1.4.8
markdown-it-py==3.0.0
MarkupSafe==3.0.2
marshmallow==3.26.1
matplotlib==3.10.0
mdurl==0.1.2
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
mypy-extensions==1.0.0
nh3==0.2.20
nodeenv==1.9.1
numpy==2.2.2
numpydoc==1.8.0
packaging==24.2
pdf2image==1.17.0
pefile==2023.2.7
pillow==11.1.0
Pint==0.24.4
pkginfo==1.12.0
platformdirs==4.3.6
pluggy==1.5.0
poetry==2.0.1
poetry-core==2.0.1
poetry-pyinstaller-plugin==1.2.1
pooch==1.8.2
pre_commit==4.1.0
propcache==0.2.1
protobuf==5.29.3
pydantic==2.10.5
pydantic_core==2.27.2
pydata-sphinx-theme==0.15.4
pyflakes==3.2.0
pygltflib==1.16.3
Pygments==2.19.1
pyinstaller==6.11.1
pyinstaller-hooks-contrib==2024.11
pyparsing==3.2.1
pyproject-api==1.9.0
pyproject_hooks==1.2.0
pytest==8.3.4
pytest-asyncio==0.25.3
pytest-cov==6.0.0
pytest-mock==3.14.0
pytest-order==1.3.0
pytest-rerunfailures==15.0
pytest-xdist==3.6.1
python-dateutil==2.9.0.post0
pytz==2024.2
pyvista==0.44.2
pywin32==308
pywin32-ctypes==0.2.3
PyYAML==6.0.2
RapidFuzz==3.11.0
readme_renderer==44.0
requests==2.32.3
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.9.4
scipy==1.15.1
scooby==0.10.0
semver==3.0.4
setuptools==75.8.0
shellingham==1.5.4
six==1.17.0
sniffio==1.3.1
snowballstemmer==2.2.0
soupsieve==2.6
Sphinx==8.1.3
sphinx-copybutton==0.5.2
sphinx_design==0.6.1
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
starlette==0.41.3
tabulate==0.9.0
tomli==2.2.1
tomlkit==0.13.2
tox==4.24.1
trame==3.7.6
trame-client==3.5.2
trame-server==3.3.0
trame-vtk==2.8.14
trame-vuetify==2.8.1
trove-classifiers==2025.1.15.22
twine==6.1.0
typing-inspect==0.9.0
typing_extensions==4.12.2
tzdata==2025.1
urllib3==2.3.0
uvicorn==0.34.0
virtualenv==20.29.1
vtk==9.3.1
websockets==15.0
wrapt==1.17.2
wslink==2.2.2
yarl==1.18.3
zipp==3.21.0
```

RyanJWard, 2025-02-26:
I am testing a fix for scdocx. The issue was related to tolerance

RyanJWard, 2025-03-07:
@MikeJanes share_topology now works and will save to scdocx and pmdb. It doesn't save to .stride.

RyanJWard, 2025-03-24:
@ans-spohekar This should be fixed now, please let us know if you are still seeing share_topology save issues

closed 2025-03-24 by RyanJWard