ansys/pyansys-geometry #1795: design.plot throws error after calling design.insert_file

ans-spohekar, 2025-03-04:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

design.insert_file fails

[image]

### 📝 Steps to reproduce

```python
from ansys.geometry.core import launch_modeler_with_core_service
modeler = launch_modeler_with_core_service()
path = FILES / "Concentric_Pipe.dsco"
design = modeler.open_file(path)
path2 = FILES / "Boxes_Shared_Topo.dsco"
design.insert_file(file_location=path2)
print(design.bodies)
design.plot()
```

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

252

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
aiohappyeyeballs==2.4.6
aiohttp==3.11.13
aiosignal==1.3.2
annotated-types==0.7.0
ansys-api-dbu==0.3.13
ansys-api-geometry==0.4.42
ansys-geometry-core==0.9.1
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.3
anyio==4.8.0
attrs==25.1.0
beartype==0.19.0
certifi==2025.1.31
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
dataclasses-json==0.6.7
Deprecated==1.2.18
execnet==2.1.1
fastapi==0.115.8
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.67.1
h11==0.14.0
idna==3.10
iniconfig==2.0.0
kiwisolver==1.4.8
marshmallow==3.26.1
matplotlib==3.10.0
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
mypy-extensions==1.0.0
numpy==2.2.3
numpy-stl==3.2.0
packaging==24.2
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pluggy==1.5.0
pooch==1.8.2
propcache==0.3.0
protobuf==5.29.3
py-cpuinfo==9.0.0
pydantic==2.10.6
pydantic_core==2.27.2
pygltflib==1.16.3
pyparsing==3.2.1
pytest==8.3.5
pytest-benchmark==5.1.0
pytest-xdist==3.6.1
python-dateutil==2.9.0.post0
python-utils==3.9.1
pyvista==0.44.2
requests==2.32.3
scipy==1.15.2
scooby==0.10.0
semver==3.0.4
six==1.17.0
sniffio==1.3.1
starlette==0.45.3
stl==0.0.3
trame==3.8.0
trame-client==3.6.0
trame-server==3.3.0
trame-vtk==2.8.15
trame-vuetify==2.8.1
typing-inspect==0.9.0
typing_extensions==4.12.2
urllib3==2.3.0
uvicorn==0.34.0
vtk==9.3.1
websockets==15.0
wrapt==1.17.2
wslink==2.3.2
yarl==1.18.3
```

RobPasMue, 2025-03-04:
@ans-spohekar -- could you clarify the title or the description please? They seem to be requesting different things.

RobPasMue, 2025-03-04:
@jonahrb - would you mind having a look at it whenever you have the chance? You might need to reach out to @ans-spohekar to get access to the files.

RobPasMue, 2025-07-14:
@RyanJWard I believe that this should no longer happen with the latest versions of Core Service - right? This is somewhat similar to what you tested in #2103

RyanJWard, 2025-07-14:
> [@RyanJWard](https://github.com/RyanJWard) I believe that this should no longer happen with the latest versions of Core Service - right? This is somewhat similar to what you tested in [#2103](https://github.com/ansys/pyansys-geometry/pull/2103)

@RobPasMue This looks very similar, I can double check this workflow

RobPasMue, 2025-07-14:
Awesome, thanks a lot. I believe it can be any 2 files - not necessarily the ones @ans-spohekar was inserting in the example