ansys/pyansys-geometry #1807: Issue calling translate_sketch_plane_by_distance when default units is not meters

craig-connelly, 2025-03-10:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

This issue is seen when I set the default units to millimeters:

DEFAULT_UNITS.LENGTH = UNITS.mm

In my code I'm translating a sketch plane by 10mm in the z direction:

sketch.translate_sketch_plane_by_distance(UnitVector3D([0, 0, 1]), Distance(10, UNITS.mm))

However the sketch plane is translated 10m in the z direction

### 📝 Steps to reproduce

Run the following code.  At the end, a plot of the extruded outline is made, in PyVista you can observe the extruded outline is extremely far away from the sketch plane in the plot (see image at end of code).  This is because the sketch plane for this extruded body was translated 10m in the z direction instead of 10mm.

# Import libraries
import os
import sys
from pathlib import Path
from pyvista.plotting import system_supports_plotting
from ansys.geometry.core import *
from ansys.geometry.core.connection import *
from ansys.geometry.core.designer import *
from ansys.geometry.core.materials import *
from ansys.geometry.core.math import *
from ansys.geometry.core.misc import *
from ansys.geometry.core.misc.units import *
from ansys.geometry.core.parameters import *
from ansys.geometry.core.plotting import *
from ansys.geometry.core.shapes import *
from ansys.geometry.core.sketch import *
from ansys.geometry.core.tools import *
import math

# Connect to Core Service backend
modeler = Modeler(host="127.0.0.1", port=50051, timeout=10)

# Create initial design and set default units to millimeters
design = modeler.create_design("bracket")
DEFAULT_UNITS.LENGTH = UNITS.mm

# Extrude bracket body
sketch = Sketch()
segment = sketch.segment(Point2D([0, 0]), Point2D([191.5319, 20]))
segment = segment.segment_to_point(Point2D([196, 10]))
segment = segment.segment_to_point(Point2D([8, -74]))
segment = segment.segment_to_point(Point2D([-16, -74]))
segment = segment.segment_to_point(Point2D([0, 0]))
body = design.extrude_sketch("bracket_extrude", sketch, 10)

# Cut three holes in the body
sketch = Sketch()
sketch.circle(Point2D([12.3649, -13.79040]), 5)
sketch.circle(Point2D([2.5899, -59]), 5)
sketch.circle(Point2D([179.5442, 11.3572]), 5)

faces = body.project_curves(direction=UNITVECTOR3D_Z, sketch=sketch, closest_face=True, only_one_curve=True)
new_edges, new_faces = body.imprint_curves(faces, sketch)
for face in new_faces:
    hole = design.extrude_face("hole", face, 10)
    body.subtract(hole)

# Sketch first cutout
sketch = Sketch()
segment = sketch.segment(Point2D([32.6723, -3.3362]), Point2D([25.422, -58.9219]))
segment = segment.segment_to_point(Point2D([47.4057, -46.7735]))
segment = segment.segment_to_point(Point2D([50.6723, -38.7762]))
segment = segment.segment_to_point(Point2D([78.8, -26.4]))
segment = segment.segment_to_point(Point2D([32.6723, -3.3362]))

# Sketch second cutout
segment = sketch.segment(Point2D([46.4086, -1.2043]), Point2D([86.4086, -21.2043]))
segment = segment.segment_to_point(Point2D([96.9864, 4.6925]))
segment = segment.segment_to_point(Point2D([46.4086, -1.2043]))

# Sketch third cutout
segment = sketch.segment(Point2D([107.5841, 5.928]), Point2D([97.7928, -18.0432]))
segment = segment.segment_to_point(Point2D([107.8933, -13.5989]))
segment = segment.segment_to_point(Point2D([118.6723, -15.4162]))
segment = segment.segment_to_point(Point2D([150.6723, -1.3362]))
segment = segment.segment_to_point(Point2D([107.5841, 5.928]))

# Sketch fourth cutout
segment = sketch.segment(Point2D([129.2148, 8.4499]), Point2D([160.8082, 3.1236]))
segment = segment.segment_to_point(Point2D([164.6723, 12.5838]))
segment = segment.segment_to_point(Point2D([129.2148, 8.4499]))

faces = body.project_curves(direction=UNITVECTOR3D_Z, sketch=sketch, closest_face=True, only_one_curve=True)
new_edges, new_faces = body.imprint_curves(faces, sketch)
for face in new_faces:
    cutout = design.extrude_face("cutout", face, 10)
    body.subtract(cutout)

# Define outline sketch
def outline_sketch(sketch):
    segment = sketch.segment(Point2D([28, -1.059]), Point2D([28, -14]))
    segment = segment.segment_to_point(Point2D([20, -58]))
    segment = segment.segment_to_point(Point2D([20, -65.5623]))
    segment = segment.segment_to_point(Point2D([55.1617, -48.8931]))
    segment = segment.segment_to_point(Point2D([57.4117, -42.8931]))
    segment = segment.segment_to_point(Point2D([102.3421, -23.7647]))
    segment = segment.segment_to_point(Point2D([108.3421, -26.0147]))
    segment = segment.segment_to_point(Point2D([164.5502, -1.8342]))
    segment = segment.segment_to_point(Point2D([170.77906, 14.8067]))
    segment = segment.segment_to_point(Point2D([28, -1.059]))

# Sketch outline on the front of the bracket
sketch = Sketch()
outline_sketch(sketch)

# Extrude and subtract front outline
outline = design.extrude_sketch("outline", sketch, 2)
body.subtract(outline)

# Sketch outline on the back of the bracket
sketch = Sketch()
sketch.translate_sketch_plane_by_distance(UnitVector3D([0, 0, 1]), Distance(10, UNITS.mm))
outline_sketch(sketch)

# Extrude and subtract back outline
outline = design.extrude_sketch("outline", sketch, -2)
outline.plot()
# body.subtract(outline)

[image]

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

25.2

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
aiohappyeyeballs==2.5.0
aiohttp==3.11.13
aiosignal==1.3.2
ansys-api-dbu==0.3.14
ansys-api-geometry==0.4.44
ansys-api-platform-instancemanagement==1.1.2
-e git+https://github.com/ansys/pyansys-geometry.git@3bef7363db276117a43c0ca71145c469f2ca5167#egg=ansys_geometry_core
ansys-platform-instancemanagement==1.1.2
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.3
anyio==4.8.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==3.0.0
attrs==25.1.0
beartype==0.19.0
beautifulsoup4==4.13.3
bleach==6.2.0
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
coverage==7.6.12
cycler==0.12.1
dataclasses-json==0.6.7
debugpy==1.8.13
decorator==5.2.1
defusedxml==0.7.1
Deprecated==1.2.18
docker==7.1.0
executing==2.2.0
fastjsonschema==2.21.1
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
fqdn==1.5.1
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.67.1
idna==3.10
importlib_metadata==8.6.1
iniconfig==2.0.0
ipykernel==6.29.5
ipython==9.0.2
ipython_pygments_lexers==1.1.1
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.2
Jinja2==3.1.6
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter-events==0.12.0
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.15.0
jupyter_server_proxy==4.4.0
jupyter_server_terminals==0.5.3
jupyterlab_pygments==0.3.0
jupyterlab_widgets==3.0.13
kiwisolver==1.4.8
MarkupSafe==3.0.2
marshmallow==3.26.1
matplotlib==3.10.1
matplotlib-inline==0.1.7
mistune==3.1.2
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
mypy-extensions==1.0.0
nbclient==0.10.2
nbconvert==7.16.6
nbformat==5.10.4
nest-asyncio==1.6.0
numpy==2.2.3
overrides==7.7.0
packaging==24.2
pandocfilters==1.5.1
parso==0.8.4
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pluggy==1.5.0
pooch==1.8.2
prometheus_client==0.21.1
prompt_toolkit==3.0.50
propcache==0.3.0
protobuf==5.29.3
psutil==7.0.0
pure_eval==0.2.3
pycparser==2.22
pygltflib==1.16.3
Pygments==2.19.1
pyparsing==3.2.1
pytest==8.3.5
pytest-cov==6.0.0
pytest-xvfb==3.0.0
pytest_pyvista==0.1.9
python-dateutil==2.9.0.post0
python-json-logger==3.3.0
PyVirtualDisplay==3.0
pyvista==0.44.2
pywin32==309
pywinpty==2.0.15
PyYAML==6.0.2
pyzmq==26.2.1
referencing==0.36.2
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.23.1
scipy==1.15.2
scooby==0.10.0
semver==3.0.4
Send2Trash==1.8.3
simpervisor==1.0.0
six==1.17.0
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
terminado==0.18.1
tinycss2==1.4.0
tornado==6.4.2
traitlets==5.14.3
trame==3.8.1
trame-client==3.6.0
trame-server==3.3.0
trame-vtk==2.8.15
trame-vuetify==2.8.1
types-python-dateutil==2.9.0.20241206
typing-inspect==0.9.0
typing_extensions==4.12.2
uri-template==1.3.0
urllib3==2.3.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
websockets==15.0.1
widgetsnbextension==4.0.13
wrapt==1.17.2
wslink==2.3.2
yarl==1.18.3
zipp==3.21.0
```

RobPasMue, 2025-03-10:
Fun bug!

RobPasMue, 2025-03-10:
Minimal reproducible bug:

```py
# Set the default units to mm
DEFAULT_UNITS.LENGTH = UNITS.mm
        
# Draw a box sketch
sketch = Sketch()
sketch.box(Point2D([0, 0]), 10, 10)
        
# Verify the sketch plane origin
assert sketch.plane.origin == Point3D([0, 0, 0])

# Translate the sketch
sketch.translate_sketch_plane_by_distance(UnitVector3D([0, 0, 1]), Distance(10, UNITS.mm))
        
# Verify the new sketch plane origin
assert sketch.plane.origin == Point3D([0, 0, 10], unit=UNITS.mm) # CURRENTLY FAILS
```

PR #1808 (2025-03-10): fix: translating sketch issues when using a custom default unit
## Description
Translation of sketches wasn't being handled properly. Now it should be consistent.

## Issue linked
Closes #1807

## Checklist
- [X] I have tested my changes locally.
- [X] I have added necessary documentation or updated existing documentation.
- [X] I have followed the coding style guidelines of this project.
- [X] I have added appropriate unit tests.
- [X] I have reviewed my changes before submitting this pull request.
- [X] I have linked the issue or issues that are solved to the PR if any.
- [X] I have assigned [... cut]
closed 2025-03-10 by RobPasMue