ansys/pyansys-geometry #1817: Handle gRPC API calls from central location

RobPasMue, 2025-03-11:
Currently, the proto calls are found all over the codebase. Looking and potential v1s, we should centralize them in some common module and handle versioning from it.

umutsoysalansys, 2025-04-15:
@RobPasMue how do we organize these v1 v2 folders, any folder structure to follow, or just place now reorganize later?

RobPasMue, 2025-04-15:
> [@RobPasMue](https://github.com/RobPasMue) how do we organize files in these v1 v2 folders, is there any folder structure to follow, or just place files there now and then reorganize later?

Hi @umutsoysalansys ! Good question, for now just focus on the v0 implementation and the v1 placeholders just like it's done for the other stubs. We will see if a v2 comes around but extending should be straightforward.

closed 2025-10-20 by RobPasMue