ansys/pyansys-geometry #1826: Stale edges left over after body subtract

craig-connelly, 2025-03-13:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

After performing a body subtract, there seems to be stale edge references left over in the design that throw an error when trying to query their properties.

### 📝 Steps to reproduce

Run the code below.  After creating a pipe model and subtracting a hole in the side of the pipe, a for loop goes over all the edges in the model and queries their length.  However, two of the edges in the model will throw errors on the length query.  The error states "The design has been closed on the backend. Cannot perform any operations on it."

#import libraries
from ansys.geometry.core import *
from ansys.geometry.core.designer import *
from ansys.geometry.core.math import *
from ansys.geometry.core.misc import *
from ansys.geometry.core.shapes import *
from ansys.geometry.core.sketch import *
from ansys.geometry.core.tools import *

#connect to Core Service backend
modeler = Modeler(host="127.0.0.1", port=50051, timeout=10)

#model pipe1
design = modeler.create_design("problem_edges")
origin = Point3D([0, -25, 0])
sketch = Sketch(Plane(origin, direction_y=[0, 0, 1]))
sketch.circle(Point2D([0, 0]), 8)
sketch.circle(Point2D([0, 0]), 10)
pipe1 = design.extrude_sketch("pipe1", sketch, -50)

#cut out hole in pipe1
origin = Point3D([0, 0, 0])
sketch = Sketch(Plane(origin))
sketch.circle(Point2D([0, 0]), 4)
pipe1_cut = design.extrude_sketch("pipe1_cut", sketch, 20)
pipe1.subtract(pipe1_cut)

#print out length of every edge in the design
print(design.name)
for body in design.bodies:
    print(body.name, "faces", len(body.faces), "edges", len(body.edges))
    for edge in body.edges:
        try:
            x = edge.length
        except:
            print("problem edge found")

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

v252

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
aiohappyeyeballs==2.6.1
aiohttp==3.11.13
aiosignal==1.3.2
ansys-api-dbu==0.3.14
ansys-api-geometry==0.4.44
ansys-api-platform-instancemanagement==1.1.2
-e git+https://github.com/ansys/pyansys-geometry.git@179686bb38f2d3c08b6778b028b3c6c4105b4f93#egg=ansys_geometry_core
ansys-platform-instancemanagement==1.1.2
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.3
anyio==4.8.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==3.0.0
attrs==25.2.0
beartype==0.19.0
beautifulsoup4==4.13.3
bleach==6.2.0
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
coverage==7.6.12
cycler==0.12.1
dataclasses-json==0.6.7
debugpy==1.8.13
decorator==5.2.1
defusedxml==0.7.1
Deprecated==1.2.18
docker==7.1.0
executing==2.2.0
fastjsonschema==2.21.1
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
fqdn==1.5.1
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.67.1
grpcio-health-checking==1.67.1
idna==3.10
importlib_metadata==8.6.1
iniconfig==2.0.0
ipykernel==6.29.5
ipython==9.0.2
ipython_pygments_lexers==1.1.1
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.2
Jinja2==3.1.6
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter-events==0.12.0
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.15.0
jupyter_server_proxy==4.4.0
jupyter_server_terminals==0.5.3
jupyterlab_pygments==0.3.0
jupyterlab_widgets==3.0.13
kiwisolver==1.4.8
MarkupSafe==3.0.2
marshmallow==3.26.1
matplotlib==3.10.1
matplotlib-inline==0.1.7
mistune==3.1.2
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
mypy-extensions==1.0.0
nbclient==0.10.2
nbconvert==7.16.6
nbformat==5.10.4
nest-asyncio==1.6.0
numpy==2.2.3
overrides==7.7.0
packaging==24.2
pandocfilters==1.5.1
parso==0.8.4
pillow==11.1.0
Pint==0.24.4
platformdirs==4.3.6
pluggy==1.5.0
pooch==1.8.2
prometheus_client==0.21.1
prompt_toolkit==3.0.50
propcache==0.3.0
protobuf==5.29.3
psutil==7.0.0
pure_eval==0.2.3
pycparser==2.22
pygltflib==1.16.3
Pygments==2.19.1
pyparsing==3.2.1
pytest==8.3.5
pytest-cov==6.0.0
pytest-xvfb==3.0.0
pytest_pyvista==0.1.9
python-dateutil==2.9.0.post0
python-json-logger==3.3.0
PyVirtualDisplay==3.0
pyvista==0.44.2
pywin32==309
pywinpty==2.0.15
PyYAML==6.0.2
pyzmq==26.3.0
referencing==0.36.2
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.23.1
scipy==1.15.2
scooby==0.10.0
semver==3.0.4
Send2Trash==1.8.3
simpervisor==1.0.0
six==1.17.0
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
terminado==0.18.1
tinycss2==1.4.0
tornado==6.4.2
traitlets==5.14.3
trame==3.8.1
trame-client==3.6.0
trame-server==3.4.0
trame-vtk==2.8.15
trame-vuetify==2.8.1
types-python-dateutil==2.9.0.20241206
typing-inspect==0.9.0
typing_extensions==4.12.2
uri-template==1.3.0
urllib3==2.3.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
websockets==15.0.1
widgetsnbextension==4.0.13
wrapt==1.17.2
wslink==2.3.2
yarl==1.18.3
zipp==3.21.0
```

umutsoysalansys, 2025-03-14:
I am tracking this issue, along with other update issues.

jacobrkerstetter, 2026-01-22:
@craig-connelly I have just run this again and I believe the error must've been fixed when we switched which boolean methods we are using in the backend. Closing this.