ansys/pyansys-geometry #1868: Named selection creation using beams is not capable of retrieving them later

RobPasMue, 2025-03-26:
Issue found coming from #1863

When we implemented the lazy-loading of the various elements inside a NamedSelection we realized that there is a bug under the following circumstances.

1. You create a NamedSelection using PyAnsys Geometry containing "beams". See test ``test_named_selection_contents`` for reference
2. NamedSelection creation goes through and, apparently, the beam got added to the NS
3. When you try to retrieve the beam using the NamedSelectionStub.Get() method... the beam does not show up
4. This causes the test to fail since it is expecting a beam to be part of its NS.

Previously, since we were not verifying this and we did not implement the lazy-loading/caching... this was not identified. But this bug is probably since the beginning.

jacobrkerstetter, 2025-03-26:
@RobPasMue Complete the Service-side PR - You can go ahead and close this if this solves your issue.

RobPasMue, 2025-03-26:
We need to revert this @jacobrkerstetter - this will verify that things are working again.

https://github.com/ansys/pyansys-geometry/blob/010ac2e79564ff1b06aefd50ba0ffa80a3942da3/tests/integration/test_design.py#L491-L496

Can you submit a PR?

closed 2025-03-31 by jacobrkerstetter