ansys/pyansys-geometry #1955: Use RPaths & env config from withing the core service binaries

ans-spohekar, 2025-05-02:
### 📝 Description of the feature

When launching core geometry service, the user need to set few env variables
ANS_DSCO_REMOTE_LOGS_CONFIG
ANSYS_CI_INSTALL
P_SCHEMA
ANSYSCL252_DIR
LD_LIBRARY_PATH

It would be good that core_service_launcher itself does these config or use RPaths

RobPasMue, 2025-05-03:
Hi @ans-spohekar - this is already the case if you use the ``launch_modeler_with_core_service``. Have you given it a try? Remove all those envirnoment variables and attempt to launch Core Service using PyAnsys Geometry.

ans-spohekar, 2025-05-05:
HI @RobPasMue 
If we fail to set `LD_LIBRARY_PATH` which points to '/core-service/Native/Linux/:/core-service/CADIntegration/bin/'
It doesn't launch the service

PR #1958 (2025-05-05): fix: core service launcher missing CADIntegration bin folder in path
## Description
Seems like ``LD_LIBRARY_PATH`` (on Linux) and ``PATH`` (on Windows) need to have ``CADIntegration/bin`` defined... Also, prepending to both ``LD_LIBRARY_PATH`` and ``PATH`` to avoid import resolution issues

## Issue linked
Closes #1955

## Checklist
- [x] I have tested my changes locally.
- [x] I have added necessary documentation or updated existing documentation.
- [x] I have followed the coding style guidelines of this project.
- [x] I have added appropriate unit tests.
- [x] I have reviewed my changes befo [... cut]
closed 2025-05-05 by RobPasMue