ansys/pyansys-geometry #1986: The geometry service can read&write files without a running license

heikeOnScale, 2025-05-15:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

On my linux machine, I get only a license server checkout, when I have got a running VPN connection.

I have got two test scripts: 1. test the license checkout by launching prime
                                           2. script that converts a dcso file into an fmd file

When I close the VPN connection, the prime license checkout fails correctly.
When I run the conversion script (dsco-fmd), then this service is still able to convert  the file even without a running license server.

[convert_scipt_and_test_data.zip](https://github.com/user-attachments/files/20226839/convert_scipt_and_test_data.zip)

[test_prime_license.zip](https://github.com/user-attachments/files/20226842/test_prime_license.zip)

                                           

### 📝 Steps to reproduce

I attached both python scripts to the issue.

1. Setup up your env to be able to run the convert_dsco_to_fmd.py script
2. then disconnect the license server
3. Rerun the convert_dsco_to_fmd.py script --> it will still convert the file

you can use the prime script to check whether you are really disconnected from an active license server (or you something else to test)

### 💻 Which operating system are you using?

Linux

### 📀 Which ANSYS version are you using?

v252

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
accessible-pygments==0.0.5
aiohappyeyeballs==2.6.1
aiohttp==3.11.18
aiosignal==1.3.2
alabaster==1.0.0
altgraph==0.17.4
annotated-types==0.7.0
ansys-api-dbu==0.3.19
ansys-api-fluent==0.3.35
ansys-api-geometry==0.4.58
ansys-api-meshing-prime==0.1.5.dev1
ansys-api-platform-instancemanagement==1.1.2
ansys-api-tools-filetransfer==0.1.1
ansys-fluent-core==0.30.5
ansys-geometry-core==0.10.7
ansys-meshing-prime==0.7.0a1.dev32
ansys-platform-instancemanagement==1.1.2
-e git+https://github.com/ansys-internal/simba-app.git@71124f3fcafe4ba858c24a707abe8d4288ee610c#egg=ansys_simba_app
ansys-simba-core==0.56.1
ansys-simba-plugin-fluent==0.17.15
ansys-simba-plugin-geometry==0.7.14
-e git+https://github.com/ansys-internal/simba-plugin-prime.git@36d0cb2772e50018d7b580e4b5609b6b1613436a#egg=ansys_simba_plugin_prime
ansys-simba-settings==0.22.4
ansys-sphinx-theme==1.3.3
ansys-tools-filetransfer==0.1.1
ansys-tools-path==0.7.1
ansys-units==0.4.0
anyio==4.8.0
appdirs==1.4.4
async-timeout==5.0.1
attrs==24.2.0
autoflake==2.3.1
babel==2.17.0
backports.tarfile==1.2.0
beartype==0.19.0
beautifulsoup4==4.13.3
build==1.2.1
CacheControl==0.14.0
cachetools==5.5.2
certifi==2025.1.31
cffi==1.16.0
cfgv==3.4.0
chardet==5.2.0
charset-normalizer==3.4.1
cleo==2.1.0
click==8.1.8
colorama==0.4.6
contourpy==1.3.1
coverage==7.7.0
crashtest==0.4.1
cryptography==42.0.5
cycler==0.12.1
defusedxml==0.7.1
distlib==0.3.8
dnspython==2.7.0
docker==7.1.0
docutils==0.21.2
dulwich==0.22.8
exceptiongroup==1.2.2
execnet==2.1.1
fastapi==0.115.12
fastjsonschema==2.19.1
filelock==3.13.4
findpython==0.6.3
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
frozenlist==1.5.0
geomdl==5.3.1
googleapis-common-protos==1.69.1
grpcio==1.68.1
grpcio-health-checking==1.62.3
grpcio-status==1.62.3
h11==0.14.0
httpcore==1.0.7
httpx==0.28.1
id==1.5.0
identify==2.6.9
idna==3.10
imagesize==1.4.1
importlib_metadata==8.7.0
iniconfig==2.0.0
installer==0.7.0
jaraco.classes==3.4.0
jaraco.context==6.0.1
jaraco.functools==4.1.0
jeepney==0.8.0
Jinja2==3.1.6
joblib==1.4.2
keyring==25.6.0
kiwisolver==1.4.8
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.1
mdurl==0.1.2
more-itertools==10.2.0
msgpack==1.0.8
multidict==6.1.0
nh3==0.2.21
nltk==3.9.1
nodeenv==1.9.1
numpy==1.26.4
numpydoc==1.8.0
packaging==24.2
pandas==2.2.3
pbs-installer==2025.3.17
pdf2image==1.17.0
pillow==11.1.0
Pint==0.24.4
pkginfo==1.12.1.2
platformdirs==4.3.6
pluggy==1.5.0
poetry==2.1.1
poetry-core==2.1.1
poetry-plugin-export==1.9.0
poetry-pyinstaller-plugin==1.3.0
pre_commit==4.1.0
propcache==0.3.0
protobuf==4.25.6
pyansys-tools-report==0.8.2
pycparser==2.22
pydantic==2.11.4
pydantic_core==2.33.2
pydata-sphinx-theme==0.16.1
pyflakes==3.2.0
Pygments==2.19.1
pyinstaller==6.12.0
pyinstaller-hooks-contrib==2025.1
pyparsing==3.2.1
pyproject-api==1.9.0
pyproject_hooks==1.0.0
pytest==8.3.5
pytest-cov==6.0.0
pytest-mock==3.14.0
pytest-xdist==3.6.1
python-dateutil==2.9.0.post0
pytz==2025.2
PyYAML==6.0.2
rapidfuzz==3.8.1
readme_renderer==44.0
regex==2024.11.6
requests==2.32.3
requests-toolbelt==1.0.0
rfc3986==2.0.0
rich==13.9.4
scipy==1.15.2
scooby==0.10.0
SecretStorage==3.3.3
semver==3.0.4
shellingham==1.5.4
six==1.17.0
sniffio==1.3.1
snowballstemmer==2.2.0
soupsieve==2.6
Sphinx==8.1.3
sphinx-copybutton==0.5.2
sphinx_design==0.6.1
sphinxcontrib-applehelp==2.0.0
sphinxcontrib-devhelp==2.0.0
sphinxcontrib-htmlhelp==2.1.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==2.0.0
sphinxcontrib-serializinghtml==2.0.0
starlette==0.46.1
tabulate==0.9.0
tomli==2.2.1
tomlkit==0.12.4
tox==4.24.2
tqdm==4.67.1
trove-classifiers==2024.4.10
twine==6.1.0
typing-inspection==0.4.0
typing_extensions==4.12.2
tzdata==2025.2
urllib3==2.3.0
uvicorn==0.34.2
virtualenv==20.29.3
websockets==15.0.1
yarl==1.18.3
zipp==3.21.0
zstandard==0.23.0
```

RobPasMue, 2025-05-15:
Thanks for raising the issue @heikeOnScale -- currently the license is checked on startup, and thus the behavior you are seeing is expected I believe. I will need to check with the team to see if we want it to continue to be this way.

RobPasMue, 2025-05-15:
Deleting the issue @heikeOnScale to avoid exposing it