ansys/pyansys-geometry #2031: Surface Body property can't get update from the backend.

umutsoysalansys, 2025-06-09:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

surface body property is set during the body object creation on the client side, but we don't have a mechanism to query that property from the backend and update it (i.e. in case there is a repair operation a surface body can be converted into a solid body) 

### 📝 Steps to reproduce

test_find_and_fix_duplicate_faces with the update_desing_from_tracker flag set to True exposes the issue. 

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
aiohappyeyeballs==2.6.1
aiohttp==3.11.15
aiosignal==1.3.2
ansys-api-dbu==0.3.18
ansys-api-geometry==0.4.57
ansys-api-platform-instancemanagement==1.1.2
-e git+https://github.com/ansys/pyansys-geometry.git@61dc9c62ae097e3c2652309d19afbfa102cd3469#egg=ansys_geometry_core
ansys-platform-instancemanagement==1.1.2
ansys-tools-path==0.7.1
ansys-tools-visualization-interface==0.8.3
anyio==4.9.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==3.0.0
async-lru==2.0.5
attrs==25.3.0
babel==2.17.0
beartype==0.20.2
beautifulsoup4==4.13.3
bleach==6.2.0
certifi==2025.1.31
cffi==1.17.1
cfgv==3.4.0
charset-normalizer==3.4.1
click==8.1.8
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
coverage==7.8.0
cycler==0.12.1
dataclasses-json==0.6.7
debugpy==1.8.13
decorator==5.2.1
defusedxml==0.7.1
Deprecated==1.2.18
distlib==0.3.9
docker==7.1.0
executing==2.2.0
fastjsonschema==2.21.1
filelock==3.18.0
flexcache==0.3
flexparser==0.4
fonttools==4.56.0
fqdn==1.5.1
frozenlist==1.5.0
geomdl==5.3.1
grpcio==1.71.0
grpcio-health-checking==1.71.0
h11==0.14.0
httpcore==1.0.7
httpx==0.28.1
identify==2.6.9
idna==3.10
importlib_metadata==8.6.1
iniconfig==2.1.0
ipykernel==6.29.5
ipython==9.0.2
ipython_pygments_lexers==1.1.1
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.2
Jinja2==3.1.6
json5==0.10.0
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter==1.1.1
jupyter-console==6.6.3
jupyter-events==0.12.0
jupyter-lsp==2.2.5
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.15.0
jupyter_server_proxy==4.4.0
jupyter_server_terminals==0.5.3
jupyterlab==4.3.6
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==3.0.13
kiwisolver==1.4.8
MarkupSafe==3.0.2
marshmallow==3.26.1
matplotlib==3.10.1
matplotlib-inline==0.1.7
mistune==3.1.3
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.3.0
mypy-extensions==1.0.0
nbclient==0.10.2
nbconvert==7.16.6
nbformat==5.10.4
nest-asyncio==1.6.0
nodeenv==1.9.1
notebook==7.3.3
notebook_shim==0.2.4
numpy==2.2.4
overrides==7.7.0
packaging==24.2
pandocfilters==1.5.1
parso==0.8.4
pillow==11.2.0
Pint==0.24.4
platformdirs==4.3.7
pluggy==1.5.0
pooch==1.8.2
pre_commit==4.2.0
prometheus_client==0.21.1
prompt_toolkit==3.0.50
propcache==0.3.1
protobuf==5.29.3
psutil==7.0.0
pure_eval==0.2.3
pycparser==2.22
pygltflib==1.16.3
Pygments==2.19.1
pyparsing==3.2.3
pytest==8.3.5
pytest-cov==6.1.1
pytest-xvfb==3.1.1
pytest_pyvista==0.1.9
python-dateutil==2.9.0.post0
python-json-logger==3.3.0
PyVirtualDisplay==3.0
pyvista==0.44.2
pywin32==310
pywinpty==2.0.15
PyYAML==6.0.2
pyzmq==26.3.0
referencing==0.36.2
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.24.0
scipy==1.15.2
scooby==0.10.0
semver==3.0.4
Send2Trash==1.8.3
setuptools==78.1.0
simpervisor==1.0.0
six==1.17.0
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
terminado==0.18.1
tinycss2==1.4.0
tornado==6.4.2
traitlets==5.14.3
trame==3.8.1
trame-client==3.6.1
trame-server==3.4.0
trame-vtk==2.8.15
trame-vuetify==2.9.0
types-python-dateutil==2.9.0.20241206
typing-inspect==0.9.0
typing_extensions==4.13.0
uri-template==1.3.0
urllib3==2.3.0
virtualenv==20.30.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.8.0
websockets==15.0.1
widgetsnbextension==4.0.13
wrapt==1.17.2
wslink==2.3.3
yarl==1.18.3
zipp==3.21.0
```

RobPasMue, 2025-06-09:
Thanks for opening the tracker issue @umutsoysalansys!

umutsoysalansys, 2025-07-15:
addressing in #2118