ansys/pyansys-geometry #2056: Clearly mark features that require unreleased Ansys versions in the documentation

Cerberus-ik, 2025-06-20:
### Description of the modifications

Hi there, in the past days I keep running into issues with my code due to some functions requiring Ansys 2025 R2 (which has not released at the time of writing).

Examples I have run into

- bounding box
- shell_body

I would prefer to have them marked as 'unusable' or 'Ansys Beta' somewhere.

Maybe in the future py_ansys functions could not be merged into the current (stable) release but rather stay in the dev build until they are usable by the public.

On this note, is it possible to gain access to Ansys beta versions? The Download portal doesn't offer anything like that and our support rep hasn't heard of anything either.

### Useful links and references

https://geometry.docs.pyansys.com/version/stable/api/ansys/geometry/core/designer/body/Body.html#Body.shell_body

RobPasMue, 2025-06-20:
Hi @Cerberus-ik! Thanks for bringing this up to my attention.

I agree.. it's probably not that clear on the documentation. These methods (in their implementation) have a wrapper that checks whether it can be used or not with your current running version... which is probably the one throwing the error you are getting. See for example:

https://github.com/ansys/pyansys-geometry/blob/6f0d394e3db5c3593fda3f6590f5f0377570875b/src/ansys/geometry/core/designer/body.py#L855-L860

We should add a note to the docstrings whenever there is a ``min_backend_version`` decorator call. So that users can know from the documentation whether a method is usable or not.

Regarding your point about the Download portal, that is true.. it is not available beforehand. The good thing is that 25R2 is right around the corner :)

Cerberus-ik, 2025-06-21:
Thank you for the quick response, I will close the issue now :)

closed 2025-06-21 by Cerberus-ik