ansys/pyansys-geometry #2099: Bug located in ...

syscordan, 2025-07-09:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

upload_file is scanning the dir and uploading all of the individual files. It looks like it wasn't coded to handle dirs. 
 
Here's the failure point
[pyansys-geometry/src/ansys/geometry/core/modeler.py at main · ansys/pyansys-geometry](https://github.com/ansys/pyansys-geometry/blob/main/src/ansys/geometry/core/modeler.py)

[image]

The loop just needs to skip dirs

### 📝 Steps to reproduce

Ensure the location the CATProduct is at has no folder in it
willnotopen = Path("C:\\Assembly\\Jet_Engine_Assembly", "Exhaust_Assembly.CATProduct")
design = modeler.open_file(willnotopen)
The model can open.
Now create a folder in the same location as the CATProduct and run the above code. You get an error about it wanting a file and not a directory. 

[Jet_Engine_Assembly.zip](https://github.com/user-attachments/files/21150071/Jet_Engine_Assembly.zip)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

261

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
import os
import sys
import math
import numpy as np
from pint import Quantity
from pyvista.plotting import system_supports_plotting
from ansys.geometry.core import *
from ansys.geometry.core.connection import *
from ansys.geometry.core.designer import *
from ansys.geometry.core.materials import *
from ansys.geometry.core.math import *
from ansys.geometry.core.misc import *
from ansys.geometry.core.misc.units import UNITS
from ansys.geometry.core.plotting import *
from ansys.geometry.core.sketch import *
from ansys.geometry.core.designer.part import *
from ansys.geometry.core.designer.part import *
from ansys.geometry.core.misc import DEFAULT_UNITS, UNITS, Accuracy, Angle, Distance, checks
import pytest
from pathlib import Path
```

jonahrb, 2025-07-10:
@RobPasMue @syscordan so i guess the next question is should we just skip over directories or recurse into them and upload all the files in there?

RobPasMue, 2025-07-10:
If there can be nested directories then... We need to recurse into all of them

syscordan, 2025-07-10:
@RobPasMue @jonahrb you should skip over them, but let me check other assembly files to see ensure that none of them have extra files that are nested in the folder. When I tested this out, my folders were empty or contained an extra copy of a file. This also happens for solidege assemblies too, but not solidworks assemblise

syscordan, 2025-07-10:
@RobPasMue you are correct. I am double checking to ensure that I find no assemblies that need the nested folders

syscordan, 2025-07-10:
@RobPasMue @jonahrb, so far all I have found where an assembly uses a nest folder for parts is JT which I testing now to see if I can open and plot the bodies

RobPasMue, 2025-07-11:
Okay - so we sync the entire directory, including nested directories

RobPasMue, 2025-07-11:
Thanks for investigating @syscordan !

RobPasMue, 2025-12-18:
Hi @ZachDerocher! This will be solved in the v1 implementation of our protofiles interface.. currently it is not set by default. Only starting on 26R1 will this feature be available.

However, since you are an Ansys employee and you might have access to the 26R1 version, feel free to give it a try. Whenever you call the ``Modeler(...)`` constructor or the ``launch_modeler(...)`` functions, make sure to define the following argument: ``proto_version="v1"``. I'm not sure if it's on the released version though... most probably not. It will only be available on the main branch currently.

I would also suggest you to reach out to @jacobrkerstetter during the winter break, he might be able to help you out on this and give it a try to your issue!

jacobrkerstetter, 2026-02-17:
@ZachDerocher I'm currently able to do this using the latest build. Can you ping me what you're seeing and if you're using the latest build