ansys/pyansys-geometry #2121: Make component independent

RyanJWard, 2025-07-16:
### 📝 Description of the feature

When a user imports a file with instanced components, it would be useful to have a make independent command to break the instancing. 

### 💡 Steps for implementing the feature

Components should have an option to make independent. Removing instancing can be useful for some modeling operations

RobPasMue, 2025-07-17:
I might need some clarity here @RyanJWard 

Say we have the following:

```bash
                                                     | -- COMP_1_BODY_1
                                      | -- COMP_1 ---|
                                      |              | -- COMP_1_BODY_2
DESIGN ---|---> INSERTED_FILE_COMP ---|
                                      |          
                                      | -- COMP_2 ---|
                                                     |
                                                     | -- COMP_2_BODY_1
                                                     | -- COMP_2_BODY_2
```

What is it you want to do? You want to flatten this into the following?

```bash
                         | -- COMP_1_BODY_1
          | -- COMP_1 ---|
          |              | -- COMP_1_BODY_2
DESIGN ---|
          |          
          | -- COMP_2 ---|
                         |
                         | -- COMP_2_BODY_1
                         | -- COMP_2_BODY_2
```

RyanJWard, 2025-07-17:
@RobPasMue In SpaceClaim if we extrude a face of an instance that same face of all instances gets extruded 

[image]

closed 2025-07-24 by jonahrb