ansys/pyansys-geometry #2149: Exposing WarpBodyByMeshNew from SpaceClaim to pyansys-geometry

dougshidong, 2025-08-01:
### 📝 Description of the feature

The WarpBodyByMeshNew command from SpaceClaim is used by Fluent's back-to-CAD feature. At the moment, it runs a Discovery script in the background using WarpBodyByMeshNew.

However the above solution is Windows-only.

Exposing WarpBodyByMeshNew will allow a Linux implementation of our back-to-CAD feature since it will remove our dependency on Discovery.

dougshidong, 2026-01-21:
@jacobrkerstetter , that was my understanding from my previous notes and discussion with @b-matteo . It's the last item in this [ADO item](https://tfs.ansys.com:8443/tfs/ANSYS_Development/Portfolio/_workitems/edit/1323462).

> Command does not exists on the Ansys Core Geometry side. 
> Needs to be converted from C# to C++ by another team (mdjanes on GitHub: Michael Janes)

Is there either a Github issue or an ADO item for me to track this?

jacobrkerstetter, 2026-01-21:
@dougshidong I'll make a ADO item for it and send you the link