ansys/pyansys-geometry #2244: Bug located in ... design.export_to_pmdb (inside Linux)

harikrushna171, 2025-09-22:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Below code is running fine in windows system. However this code is not running inside Linux environment. 

I want to use this geometry further in Mechanical. As Discovery or spaceclaim geometry is not supported inside Linux so I used geometry service and use .pmdb format but I cannot export .pmdb file.  

### 📝 Steps to reproduce

# Start a modeler session
modeler = launch_modeler()
print(modeler)

design_name = "ExtrudeProfile"
design = modeler.create_design(design_name)

sketch = Sketch()
sketch.circle(Point2D([10, 10], UNITS.mm), Quantity(10, UNITS.mm))
cylinder = design.extrude_sketch("SingleBody", sketch, Quantity(10, UNITS.mm))

pmdb_path = design.export_to_pmdb(location=geometry_path)
modeler.close()

**error** "
raise GeometryExitedError(
ansys.geometry.core.errors.GeometryExitedError: Geometry service connection terminated: The AnsysSCPlugin.PluginAddin needs to be loaded in order to export to PMDB format.
Stack trace:
   at Infrastructure.v0.Infrastructure.ApiGeometryCoreService.Designs.PMDBExport(String filepath)
   at Infrastructure.v0.Infrastructure.ApiGeometryCoreService.Designs.<>c__DisplayClass26_0.<Export>b__0()
   at SpaceClaim.Api.V252.WriteBlock.ExecuteTask(String text, Task task)
   at Infrastructure.v0.Infrastructure.ApiGeometryCoreService.Utilities.ApiUtilities.ExecuteTask(String commandName, Task task, Metadata metadata)"

### 💻 Which operating system are you using?

Linux

### 📀 Which ANSYS version are you using?

25R2, With ANSYS Python Manager installation

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
aa
```

RobPasMue, 2025-09-24:
Hi @harikrushna171 -- thanks for opening this issue!

Could you elaborate a bit more on the following points?

* Which Ansys product are you connecting too? Is it The Geometry Service (based on Linux)?
* Can you confirm that the GeoemtryService is installed on your machine?`- Wherever your Ansys products are installed, there should be a folder called "GeometryService"
* What version of PyAnsys Geometry are you using? (i.e. ``python -c "from ansys.geometry.core import __version__;print(__version__)``)

harikrushna171, 2025-09-24:
hi, @RobPasMue , thank you for connecting. 

- ANSYS 25R2, Geometry Service running on Linux. Everything works well up to the Plot of cylinder command.
- Previously, it showed the error "Geometry services is not initialized," but now it runs without issues and I can view the cylinder. Please see attached image 

[image]

-     It is 0.11.0

harikrushna171, 2025-10-03:
Hello, @RyanJWard ,

Thank you for support and sorry for late reply,

I setup "ANSYS_GEOMETRY_SERVICE_ROOT=/my_path/to/core_geometry_service" and it is working fine. As show in above image i can launch geometry service. 

However, "design.export_to_pmdb" this specific command is not working. 

"v252/GeometryService/Logs" I dont have Logs file.

RobPasMue, 2025-10-03:
It could be a license server assignment issue as well -- @harikrushna171 could you confirm that you are storing your license server IP/address under the environment variable ``LICENSE_SERVER``, ``ANSRV_GEO_LICENSE_SERVER``, or ``ANSYSLMD_LICENSE_FILE`` (this is the precedence order BTW)

RobPasMue, 2025-10-03:
@harikrushna171 -- could you do the following change to your code?

```py
pmdb_path = design.export_to_pmdb()
# --- START ---
logfile = modeler.get_service_logs(all_logs=True, dump_to_file=True)
print("File available at:", logfile.absolute())
# --- END ---
modeler.close()
```

This will dump your server logs to a specific file.

RobPasMue, 2025-10-06:
@RyanJWard - do you have any idea why the PMDB export might be failing?

RyanJWard, 2025-10-06:
@harikrushna171 Can you also run: ls -ld /v252/GeometryService

RobPasMue, 2026-03-16:
Hi @harikrushna171 - thanks for letting us know. @RyanJWard could you please have a look into this?

In the meantime, could you follow the same changes we performed throughout this issue on your 261 installation and confirm that it works again if you do that?